python-microscope/microscope #25: Pvcam flavour

ColinBrosseau, 2017-09-11:
I'm interrested in the pvcam part of your project. I've not found what is the version you use. I believe that there are few different pvcam libraries: Photometrics, Princeton Instruments.

I got a
AttributeError: function 'pl_cam_register_callback' not found
Looking in the Pvcam 2.7 user manual, I don't see that function.

Regards

iandobbie, 2017-09-11:
Colin Brosseau <notifications@github.com> writes:

> I'm interrested in the pvcam part of your project. I've not found what is the version you use. I believe that
> there are few different pvcam libraries: Photometrics, Princeton Instruments.

Photometric and PI are basically different arms of the same company,
however different cameras can require older versions of pvcam to get
them to work. 

> I got a
> AttributeError: function 'pl_cam_register_callback' not found
> Looking in the Pvcam 2.7 user manual, I don't see that function.
>

This function is from newer versions of pvcam. Depending on your camera
you might be able to use a newer version. We have tried to drive and old
casscade II through microscope and gave up as the work necessary would
have been quite extensive. This required pvcam 2.7.

Ian

closed 2017-09-11 by ColinBrosseau

mickp, 2017-09-11:
Hi, Colin.

To add to what Ian said, I wrote the pvcam module using version 3.1 of the
SDK. I used callbacks to fetch data from the camera, as I saw stability
issues when repeatedly polling an Evolve 512, so the pvcam module only
supports more recent cameras that support callbacks.

Best regards.

-Mick.