python-microscope/microscope #30: deviceserver problems under Windows

mickp, 2018-03-21:
The deviceserver process management works well on my workstation, but fails to shutdown child processes on my laptop. Also, the use of the signals module means that the deviceserver can't be wrapped behind a windows service (which seems to be the only way to have scripts start automatically without logging in): doing so raises the error "signals only works in the main thread" (although querying the thread name at that point , it says it *is* the main thread). We probably need to use separate event types under UNIX (signal) and Windows (win32event).

iandobbie, 2020-05-05:
This must be solved as we are using microscope routinely on windows and linux.

closed 2020-05-05 by iandobbie

carandraug, 2020-05-06:
For future reference, looking into the logs, I'm guessing this was fixed in 28bda20a8040 .