python-microscope/microscope #36: support for automatic focus

carandraug, 2018-06-08:
Requested at ELMI 2018.

Hardware that performs automatic focus control. No code on our side, just have to control when that device should run so it does not happen in the middle of acquisition.

carandraug, 2018-06-09:
Ian Dobbie wrote:
> some hardware autofocus systems just need a ttl to suppress them during acquisition.  This is already possible but might require some experiment options to generate the correct signals.

carandraug, 2020-01-09:
It's possible that auto focus is actually done by software, so such device might actually be a composite device that makes use of a camera and stage.

NickHallONBI, 2020-01-09:
> It's possible that auto focus is actually done by software, so such device might actually be a composite device that makes use of a camera and stage.

Is that not basically what we're already doing with the AO as a composite device in place of a stage?