python-microscope/microscope #38: support for Hamamatsu cameras

carandraug, 2018-06-08:
Requested at ELMI 2018.

I'm told that Hamamatsu dCamAPI library supports all of their cameras.

carandraug, 2019-11-08:
Requested again, twice, at SPAOM 2019.

carandraug, 2019-12-19:
Met with @maazevedo who requested support for an Hamamatsu Orca flash 4.0.  I looked into [dcam-api](https://dcam-api.com/structure/) who claims to support both Windows and Linux but downloads only shows windows support. I have emailed Hamamatsu about where I can find the Linux builds.

carandraug, 2019-12-20:
MesoSPIM (GPLv3) have a [ctypes wrapper to dcam-api](https://github.com/mesoSPIM/mesoSPIM-control/blob/master/mesoSPIM/src/devices/cameras/hamamatsu/hamamatsu_camera.py).

carandraug, 2020-01-08:
I have just spoken with [James R Brown from Hamamatsu](mailto:JRBrown@hamamatsu.co.uk) at QBI2020 who offered to lend us their cameras for a few months if we need them to do this.

juliomateoslangerak, 2021-06-28:
Had a visit from Hamamatsu representative last week and he mentioned that there will be a Python interface for their cameras very soon.

carandraug, 2021-09-17:
@ARCFaria got an Hamamatsu camera is planning on doing this over the next weeks. @kasasxav was also interested to test the use of Microscope as hardware layer for ImSwitch.

kasasxav, 2021-09-17:
Yes! We use a similar [ctypes-wraper to dcam-api](https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imcontrol/model/interfaces/hamamatsu.py) which we borrowed from [Zhuang Lab / Babcock Lab](MIT license - https://github.com/ZhuangLab/storm-control), and then we have an intermediate class [Hamamatsu Manager](https://github.com/kasasxav/ImSwitch/blob/master/imswitch/imcontrol/model/managers/detectors/HamamatsuManager.py) to translate that into the functions we use (crop, getLatestFrame, etc).

I think something similar can be delegated to Python-Microscope and then we can directly fetch it from there. We also have Hamamatsu cameras so we debug experimentally.

carandraug, 2022-12-23:
It took a while but it's now merged. It was only tested on Linux and with software triggers though. Closing.

closed 2022-12-23 by carandraug