python-microscope/microscope #40: support hardware from old commercial microscopes

carandraug, 2018-06-08:
Requested at ELMI 2018

Users that have old commercial microscope that are no longer maintained, would be interested on using since it gets more difficult to use the old acquisition software.  For example, a LSM 510.

mickp, 2019-05-21:
I agree with Ian - this looks like a lot of work for what I expect would be a small number of users, many of whom are probably already well supported by Micro Manager.

iandobbie, 2023-03-22:
An interesting idea but far beyond the current scope of the project, especially with the specific example of a commercial laser scanning confocal. These system have a lot of custom hardware and no available documentation of the low level control of this hardware. This would be a massive reverse engineering job.

closed 2023-03-22 by iandobbie