python-microscope/microscope #41: support for lightsheet microscopes

carandraug, 2018-06-08:
Requested at ELMI 2018

Don't forget lightsheet microscopes.

carandraug, 2019-11-08:
Two groups, interested in using microscope and cockpit, also for their lightsheet microscope (group in Champalimaud (LIsbon) and i3s (Porto)).

iandobbie, 2019-11-08:
What does support for light sheet microscopes mean?  Overall is there anything different from conventional systems?  I guess the motion composting controls will need adaption but the hardware is very similar so someone just needs to build one!

Maybe we should try to get someone to build an OpenSPIM using microscope/cockpit.

Ian

carandraug, 2019-11-08:
We need to sync galvos with the cameras (well, the galvos trigger the camera). Their stage does rotation too (on the cockpit side, I will open an issue for that there, the mosaic view does not make a lot of sense)

iandobbie, 2019-11-08:
Galvo to camera timing would be an experiment on the hardware timing system.

The mosaic could become a secondary window but even in a light  sheet an overview image can be useful.

Ian

carandraug, 2019-11-08:
Christian Eggling also is interested on using this for a lightsheet microscope they plan on building.