python-microscope/microscope #42: support for galvo scanners

carandraug, 2018-06-08:
Requested at ELMI 2018.

Add support for galvo scanners.

carandraug, 2018-06-09:
Ian Dobbie wrote:
>  galvos basically take analogue so the redpitya should already be able to do this, although performance might be an issue.

carandraug, 2023-07-06:
Adding context to @mihaeladuta comment. They're working with Martin Booth and Patrick Salter who are only interested in controlling the galvo mirrors in the context of acquiring an image with a single pixel detector. I met with them earlier and I guess it makes sense to model it as a `Camera` device (maybe we can discuss this more at the Microscope development meeting  next week).

iandobbie, 2023-07-07:
I have thought about this some and the biggest issue I foresee is pixel size. Most of the obvious galvo settings can be quite easily translated into camera type settings, eg x and y pixel number, exposure time,  but typically galvo scanners on microscopes have a zoom which affects the range of scan and combined with the numbers of pixels sampled and objective gives a pixel size. 

There are already a few issues in cockpit that would benefit from the camera knowing what the pixel size is (or should be), eg
https://github.com/microscope-cockpit/cockpit/issues/629
https://github.com/microscope-cockpit/cockpit/issues/538
https://github.com/microscope-cockpit/cockpit/issues/391
https://github.com/microscope-cockpit/cockpit/issues/448

Possibly others, but this seems like a feature that needs to be implemented and properly interface this data between cockpit and microscope.

iandobbie, 2023-07-11:
This was discussed at length during the July 2023 development meeting. 

There is also an existing issue here, #152 as well as the cockpit ones above.