python-microscope/microscope #43: support for arduino via serial

carandraug, 2018-06-08:
Requested at ELMI 2018

Support for Arduino control over serial. Not sure what this would mean, I think we already do, we have a SerialDeviceMixin to mix with whatever type of device the arduino would be.

mickp, 2019-05-21:
I'm not sure I understand this one. We can already talk to serial devices, but we'd need a defined interface (not to mention defined behaviour) on the Arduino end.

beniroquai, 2020-12-06:
I think it would be valuable to have a template (```.ino``` file) which can be used for a serial communication between the microscope and the arduino. I would be happy to help and I'm also working on something. Did you work on something similar in the meantime already?

iandobbie, 2022-05-18:
To keep up with the average ~18 month reply time for this thread. No as far as I know no one has made any effort to use an ardiuno with microscope. What do you want it to do? 

I have a module that does digital IO and has a temperature measurement via i2c which runs on a raspberry Pi. I plan to port this into python-microscope shortly. The main current holdup is sitting down and deciding on sensible ABC functions.

beniroquai, 2022-05-18:
Haha, nice! Thanks :) 
In the meantime I switched from Arduino to ESP32 (due to DAC, larger Memory and wait for it: Dual core!). I created a small REST interface around it to use it via USB-Serial and Wifi. I think it would be easy to write a wrapper for the microscope framework. Do you think microscope could benefit from it?
The sources are here: https://github.com/openUC2/UC2-REST => ESP32/main

iandobbie, 2022-05-18:
Certainly sounds interesting and I always planned to try the UC2 stuff, just never got around to it. I'll have a look at the repository and have a think. I am meant to be teaching microscopy to the grad students this autumn, maybe that is a good time to try it.

beniroquai, 2022-05-19:
Certainly! :) 
In case you're looking for parts, we are "about to scale up the production"..not sure what this means :D 
The UC2-Rest is just a weird approach to standardize all the electronics in one framework. Modularity is nice - but I'm not a programmer ;-)

iandobbie, 2023-03-07:
Doubling my responce frequency to 9 months! I have now implemented both a digital IO component and a value logger to accept scalars from sensors, so far both have simulations and concrete implementations on a raspberry pi. 

Seem though that the esp32 code is designed to run the whole U2C system, so would also include stage, light source etc.. The best way to integrate this would be as a controller, to broker the connection and then we can add support for the various components.