python-microscope/microscope #45: data with device specs

carandraug, 2018-06-22:
We need something where we store the specs we use to implement the interface to the devices we have. We already have a directory shared between us in micronadmin. However, that won't work if there's people outside micron working on microscope. Also, I'd prefer that people can't accidentally remove files.

mickp, 2018-06-24:
If you're talking about the vendor-supplied documents on micronadmin, we
need to be very careful - much of that documentation is provided under
licence, and the licence may not allow us to redistribute.

carandraug, 2022-12-17:
This is very annoying because getting the vendors to give us access to the documentation is difficult and access to a specific version is just impossible.  And access to the documentation is fundamental to the development of this project.  But Mick is right, we simply cannot redistribute the documentation, so I'm closing this issue.

If we could, I would have proposed we create a git repository and dump all the documentation there with the versions used for development. We need the different versions so we can reason why something was done in that way in the past.  This would lead to a very large repository. Not only would it be a pain to clone, it would go over our traffic limit very quickly.  So I'd suggest using git LFS (and not git annex before github does not support it) which enables downloading only certain files (although that's not the default). Documentation would be not only the manuals but also any header files.

closed 2022-12-17 by carandraug