python-microscope/microscope #49: decoupling interface to devices from Pyro

carandraug, 2018-07-19:
We currently wrap all device classes with `Pyro.expose`. We don't have any secret or unsafe methods in our devices, our private methods are only to signal that a method is not for public use. We could just configure the Pyro daemon to expose the methods by default.

This change would mean:

1. we no longer need to add Pyro code to all device modules and we no longer need to remember
2. users are free to pick whatever method they want for RPC

We have some `Pyro4.behaviour('single')` and `Pyro4.oneway` wrappers for the camera, not sure how to handle those.

carandraug, 2019-03-26:
Spoke with @mickp  and pushed a change (708c7b5) that starts this by removing all `Pyro4.expose` and disables Pyro4 `REQUIRES_EXPOSE`. If we really need to prevent something fro being exposed, we can still use a leading underscore on the attribute.

carandraug, 2019-07-03:
I was earlier asked about using microscope without Pyro4. They already have their own RPC system to do the remote stuff but are interested on using our interface. I think the main place where we have is on the client for `DataDevice`.

We don't have a defined interface for such client on microscope side. If we do, and if autoproxying two-way works (not sure if it does, may be one-way only), then we could remove Pyro from there as well.

carandraug, 2019-11-04:
Some classes were decorated with `@Pyro4.behavior(instance_mode='single')`. This is the same as creating the instance and register it with the Pyro daemon so I removed those.

The pvcam module decorates the `soft_trigger` as oneway. This basically calls soft_trigger without blocking. However, the other cameras that have `soft_trigger` do not have such annotation, so maybe it's not actually necessary?

iandobbie, 2023-03-03:
I think this is now a closed issue. We just expose everything unless it start with a _ . If we wish to further pursue the idea of changing how the DataDevice sends data back to the server then this is a separate issue I think. I am going to close this as completed.

carandraug, 2023-03-14:
Reopening as this is not yet fixed. Pyro4 is still imported at top level in pvcam, for annotating the `soft_trigger`. We also import it at top level in the abc module to handle Pyro exceptions there.

iandobbie, 2023-03-22:
Oh and the DataDevice interface is simple, it probably just needs to be documented. The remote system (eg cockpit) sets up a Pyro server at "uri" and calls client=receiveClient(uri). This sets up the Pyro connection and the data is then returned by calling client.receiveData(data)