python-microscope/microscope #50: Queuing multiple sets of patterns on alpao causes failure

NickHallONBI, 2018-07-23:
Discovered this error in the course of testing something else. If you queue up a set of patterns with the queue_patterns() function and then queue up another set of patterns without clearing the first lot you get the following error:

"Traceback (most recent call last):
    File "(stdin)", line 1 in (module)
    File "C:\Python\lib\site-package\Pyro4\core.py", line 186 in __call__ return self.__send(self.__name, args, kwargs)
    File "C:\Python\lib\site-package\Pyro4\core.py", line 470, in _pyroInvoke raise data
Exception: Time limit exceeded during the transfer.(error 258)"

Additionally, queuing any subsequent patterns after this error results in the following error:

"Traceback (most recent call last):
    File "(stdin)", line 1 in (module)
    File "C:\Python\lib\site-package\Pyro4\core.py", line 186 in __call__ return self.__send(self.__name, args, kwargs)
    File "C:\Python\lib\site-package\Pyro4\core.py", line 470, in _pyroInvoke raise data
Exception: Send failed
PCIS-DASK::ERR-201: The configuration API failed.(error 260)"

David has a fix for this.

carandraug, 2018-07-23:
This only happens with hardware triggering:

```
import numpy
from microscope.devices import TriggerType, TriggerMode
from microscope.mirror.alpao import AlpaoDeformableMirror

mirror = AlpaoDeformableMirror('BIL103')
mirror.set_trigger(self, TriggerType.RISING_EDGE, TriggerMode.ONCE)
patterns = numpy.zeros((2, mirror.n_actuators))
mirror.queue_patterns(patterns)
mirror.queue_patterns(patterns) # second call makes it crash
```

Solution is to call `asdkStop` (not yet wrapped) before `asdkSendPattern`.

carandraug, 2020-08-11:
Fixed with 6b2b5bf

closed 2020-08-11 by carandraug