python-microscope/microscope #53: Andor zyla. No images.

juliomateoslangerak, 2018-07-31:
After some dll-hell I managed to reconnect the zyla using microscope. 
Everything looks fine but there are no images

Server side config:
device(AndorSDK3, '10.6.19.30', 8001, timeout=1)

Server side STOUT:
2018-07-31 17:19:52,178:__main__:INFO:PID 4704: 4668 is alive.
2018-07-31 17:19:55,298:AndorSDK3:DEBUG:PID 4668: Debugging messages on.
2018-07-31 17:19:57,185:__main__:INFO:PID 4704: 4668 is alive.
2018-07-31 17:20:02,193:__main__:INFO:PID 4704: Aggregating reps. of: 4668 is alive.
Object <microscope.cameras.andorsdk3.AndorSDK3 object at 0x0000000005824B00>:
2018-07-31 17:20:14,985:AndorSDK3:INFO:PID 4668: Device initialized; starting daemon.
    uri = PYRO:AndorSDK3@10.6.19.30:8001
Pyro daemon running.
2018-07-31 17:20:27,231:__main__:INFO:PID 4704: 5 times: 4668 is alive.

Cockpit side config:
[server]
ipAddress: 10.6.19.11
...
[Zyla-Cam1]
type: MicroscopeCamera
uri: PYRO:AndorSDK3@10.6.19.30:8001
dye: Reds
wavelength: 640
triggerSource: cRIO
triggerLine: 4
...

After enabling:
2018-07-31 17:24:13,744:AndorSDK3:DEBUG:PID 4668: Enabling ...
2018-07-31 17:24:13,744:AndorSDK3:DEBUG:PID 4668: Preparing for acquisition.
2018-07-31 17:24:13,744:AndorSDK3:DEBUG:PID 4668: Purging buffers.
2018-07-31 17:24:13,744:AndorSDK3:DEBUG:PID 4668: Creating 32 buffers.
2018-07-31 17:24:14,960:AndorSDK3:DEBUG:PID 4668: Acquisition enabled: True.
2018-07-31 17:24:14,960:AndorSDK3:DEBUG:PID 4668: ... enabled.
2018-07-31 17:24:14,960:AndorSDK3:INFO:PID 4668: Setting client to PYRO:mui@10.6.19.11:7702.

After changing exposure time:
2018-07-31 17:26:58,059:AndorSDK3:DEBUG:PID 4668: Disabling acquisition.
2018-07-31 17:26:58,137:AndorSDK3:DEBUG:PID 4668: Set exposure time to 0.010000, resulting framerate 50.297763.
2018-07-31 17:26:58,137:AndorSDK3:DEBUG:PID 4668: Preparing for acquisition.
2018-07-31 17:26:58,308:AndorSDK3:DEBUG:PID 4668: Acquisition enabled: True.

I get all the options but no images. Experiment is running through my executor, trigger is triggering but no images.

How can I access the camera through the terminal window and try some software triggers?

juliomateoslangerak, 2018-07-31:
Test camera through PYRO is working. Same server.

mickp, 2018-07-31:
First thought - is the trigger correctly configured? By default, microscope
defaults to software triggering, so you'll need to use the settings
interface in cockpit to change this,

mickp, 2018-08-01:
> I see the all the settings with the settings interface of cockpit.
> trigger_mode is indeed Software but it is grayed out and I cannot change
> it. There are other significant settings that I cannot change, like ROI. I
> can actually change:
> transform
> readout_mode
> use_callback
> num_buffers
> auxiliary_out_source
> event_enable
> event_selector
> exposed_pixel_height
> exposure_time
> external_trigger_delay
> fan_speed
> frame_rate
> io_invert
> io_selector
> spurious_noise_filter
> static_blemish_correction
> [image: image]
> <https://user-images.githubusercontent.com/7799110/43507079-61ed9e6c-956c-11e8-8db4-97f8c7cfbe74.png>

Right. The greyed out options are locked because the camera is armed and
waiting for a trigger - these are settings that the vendor library will not
let you change while the camera is in this state. Disable the camera,
change trigger mode, re-enable.

juliomateoslangerak, 2018-08-01:
I see. I was fixed in the previous version where cockpit will disarm the camera in order to change those settings.
BTW: Is that option still available for frequently changed items?

So now, when I disable the camera I get access to the items but:
If I try to change I get 'Error in subscribed func updateGrid in cockpit.gui.device.' and parameters do not change upon OK click and window freezes upon 'apply' click.

juliomateoslangerak, 2018-08-01:
cockpit/devices/microscopeCamera.py line 92
self.proxy.update_settings(settings) gets called with the right (changed) settings but in line 93, self.proxy.get_all_settings() returns the old unchanged settings... Is that expected?

I'm checking further server side

iandobbie, 2018-08-01:
If you set trigger mode to software trigger in cockpit it should work for snaps but not experiments.

Ian

uri = PYRO:AndorSDK3@10.6.19.30<mailto:AndorSDK3@10.6.19.30>:8001

uri: PYRO:AndorSDK3@10.6.19.30<mailto:AndorSDK3@10.6.19.30>:8001

2018-07-31 17:24:14,960:AndorSDK3:INFO:PID 4668: Setting client to PYRO:mui@10.6.19.11<mailto:mui@10.6.19.11>:7702.

mickp, 2018-08-01:
The settings grid won't change the camera's enabled/disabled state. It
disables the options that it can not change, depending on the camera state;
in the case of the Zyla, this is determined by a call to the SDK funciton
IsWriteable for each setting.

I found the problem in updateGrid. It forces conversion of a long int to an
int, because at some point, passing longs to a wx PropGrid
Property.SetValue crashed it. The long conversion only makes sense in
python 2, and breaks python 3. It looks like the wx issue is fixed now,
though, so I've removed the two lines and pushed it to master
(32e40bd01a64fce7ab04a32024a4847d4aa0f3f6).

juliomateoslangerak, 2018-08-01:
I'm not sure this is the problem. When looking into the server, update_setitngs is called, but no changes are applied cause, for some reason, those settings are tagged as read_only. self.settings[key]['readonly'] returns True. for those settings comming directly from the SDK.
For the case of the nr of buffers, that is defined somewhere else, readonly is false and I can set the values as expected in cockpit.

juliomateoslangerak, 2018-08-01:
I think I know what the problem might be. 
Camera connection is opened in andorsdk3 initialize (line 327) and not in SDK3Cam Init (line 150). That means that connectProperties is not called and ATProperties are not 'connected' to the right IsImplemented, IsReadable,... 

I'm trying out now

juliomateoslangerak, 2018-08-01:
@iandobbie the snap is not working either. Might this be cause it's the fpga handling the takeImage? In fact the FPGA is triggering when I snap.
 And for what I thought it was, nope. All the settings seem to be correctly constructed with a lambda function at the read_only. This function returns the right thing, at least at construction time.

juliomateoslangerak, 2018-08-02:
I found the problem. Here (microscope.devices line 295)
self.settings[key]['readonly'] should be called. I noticed that you already implemented a function for this.

Now I can change parameters and get images. I earned the right to a Thorlabs sweet!!

closed 2018-08-02 by juliomateoslangerak