python-microscope/microscope #56: Zyla 4.2plus not working

juliomateoslangerak, 2018-08-09:
My zyla Plus is not working. 
I have a classic zyla 4.2 working and the plus version that does not work under microscope. 
I have tested the last Andor Sdk without success. Will write details later. 
I have also tested connecting the zyla + to the same workstation where the zyla classic is connected. The Sdk there should just be supporting zyla +.
Have you tested the plus version? If yes, which sdk version?

mickp, 2018-08-09:
We haven't tested the Zyla plus.

juliomateoslangerak, 2018-09-26:
The camera seems to have problems with Andor's own SDK3 example code. SO presumably there is a problem with the camera, cooling or connection cables. 
I'm closing this issue and following up with Andor.

closed 2018-09-26 by juliomateoslangerak

mickp, 2018-09-26:
Check that the lane width on the PCIe slot you're using matches that
required for the CameraLink card, especially if you see the camera work
fine at lower bandwidths (i.e. with smaller image sizes and/or reduced bit
depth), but it fails for full frame capture. On some slots, motherboard
manufacturers often use physical connectors for a greater lane-width than
that slot can really support.