python-microscope/microscope #59: state of laser device after constructor

carandraug, 2018-08-28:
What should the state of the laser after the constructor? Should it be left on whatever state it was before or should it be turned off? At the moment, the constructor makes no change but for the laser sapphire it would mean that it's left on (the laser is on as soon as the key switch is turned on).  What about after `initlize()` ?

Also, should the constructor success if the key switch is off or interlock is open? Or should it fail during `initialize`?

mickp, 2018-08-28:
Do you mean the Sapphire (which I'm less familiar with) or the Cobolt
(whose interlock key is not really an interlock at all)?

Changing the state of the hardware on connection might be considered an
unintended side effect. In some cases, this is unavoidable, e.g. certain
cameras where opening the comms channel via the vendor library resets the
device because it uploads firmware, or something. This shouldn't be the
case for a serial device, though.

I think Initialize should just establish comms and return OK regardless of
interlock state. You might initialize just fine, then trip the interlock
later, so the error should come when we check that any change of power
level or emission state happens as expected.

carandraug, 2018-08-28:
> Do you mean the Sapphire (which I'm less familiar with) or the Cobolt (whose interlock key is not really an interlock at all)?

I mean all.  I mean what should the behaviour in general be?

I mentioned Sapphire because when you turn its key on, it starts emitting light. The only way to then turn the light off is to connect to it and send the command to turn the light off. I would guess this is not ideal. It is also the opposite of the behaviour of Cobolt.

When do we expect users to be turning the key switch on? Before starting the device server? Before starting whatever program that will make use of the device? After that program has grabbed the device?

> Changing the state of the hardware on connection might be considered an unintended side effect. [...] This shouldn't be the case for a serial device, though.

But we do change state. In the Sapphire we turn off the prompt mode during construction. On the Cobolt we turn off autostart during construction. We also don't return to the previous state after destruction since the destructor calls `shutdown` (which turns off the laser). This means that construction + destruction will change state, even without having ever called `initialize`. In addition, the Cobolt on destruction will also change the "on after interlock" setting (the `@cob0` command) independently of its state before construction, and we do so because we think is safer for the user.

When I opened this issue, I was leaning that after construction (or at least after `initialize`), the laser should be left in a well defined state.  When it comes to most obvious state (emitting or not emitting light) the current state is:

* Deepstar does nothing until `enable` or `disable` is called
* Cobolt turns the laser off during `initialize` (side effect of `@cob1`)
* Sapphire turns the laser off during constructor (mea culpa, until yesterday it did nothing until `enable` or `disabled` was called)

This "issue" is just about deciding what behaviour we should have.

> I think Initialize should just establish comms and return OK regardless of interlock state. You might initialize just fine, then trip the interlock later, so the error should come when we check that any change of power level or emission state happens as expected.

Yeah, I agree that we still need to handle the interlock after initialization. I just find it odd that after initialize the laser may not actually be ready to be enabled. I'm ok either way, I just want to have the expected behaviour defined.