python-microscope/microscope #62: Suggestion of stage control functions.

iandobbie, 2018-09-17:
May thoughts on a stage Api are. 

Number of axes
Name of axes

Then each axis needs 

Hardware range 
Soft limits
Home
Move relative 
Move absolute
Speed (maybe accl and decl rate)
Get position
Time to move x

Can’t think of much else 

We could have a generic serial device which then maps the calls to
specific strings as most of the stages are just serial devs with text
control strings.

Ian

carandraug, 2019-03-27:
Here's @mickp notes about coming up with a stage interface:

> Many stages use similar serial commands, but different stages have different sets of limitations, so there are some technicalities which we should think carefully about before we commit to an interface.
>
> * How are multi-axis stages treated compared to those composed from discrete-axis devices?
>
> * How to determine and respond to how multi-axis stages handle their multiple axes.
>
>   + How many and what combination of its axes can a controller move simultaneously?
>
>    + Does hitting a limit on a single axis throw an error condition for further moves on all axes, or just that one?
>
> * What do we even do about error conditions?
>
>    + How do we merge the different types of error state (e.g. hard limit, soft limit, overcurrent detection, encoder error, comms error, servo target error, servo instability), that different stages give into a manageable and meaningful set within microscope?
>
>    + How do we deal with those controllers that put themselves into a different condition on certain errors?
>
>    + Do we attempt to recover from error conditions and, if so, how do we determine that attempting to recover from a given error condition for a given stage is a safe thing to do?
>
> * How do we determine when an axis has stopped moving on a device that don't give reliable status, and has significant noise on its position readout (e.g. Linkam)?

iandobbie, 2019-05-21:
Have have an initial working version for the prior ProScan III, but it is based on Mick's latest microscope changes so I wont add a pull request till Mick's stuff is pulled.

iandobbie, 2019-05-21:
I now have a fully working implementation of the prior proScanIII controller and stage, along with an abstract class with a list of methods to be defined. It is currently in Mick's head branch as it has a large imaging area so needs the updated microscope and cockpit. Once Mick's changes are pulled I will create a pull request against it.

carandraug, 2019-07-09:
At MMC 2019, it was requested support for rotating stages.

carandraug, 2019-08-08:
We had a meeting today and discussed this. The discussion veered towards the whole issue of compositing devices and support of microscope controllers that control not only a stage but also other devices. And again about having different interfaces like discussed to implement the Toptica iChrome multi-laser engine.

Anyway, @carandraug will start with @iandobbie initial implementation of the prior ProScanIII . We will start by having it as a controller class that returns a list of its sub devices. This will be the individual stage motor and the filterwheel.

carandraug, 2020-01-24:
Proposal on my "stage-abc" branch.  This is pretty simple interface and it's expected that we either extend it later or that it is used to construct more friendly interfaces, some sort of space manager, that will require a more complex configuration.

iandobbie, 2020-05-05:
I think this is now implemented and being used so the issue should be closed. The space manager idea is a separate issue and should be opened as such.

closed 2020-05-05 by iandobbie