python-microscope/microscope #66: option to control behaviour of test camera

carandraug, 2019-03-07:
It would be useful to have test cameras with different behaviours. Currently, they only generate random images but in cockpit they can generate a wave and rotate the image each time. If this was an option when constructing the `TesCamera`, we could have:

* random : random values in all pixels
* black: all values zero
* white: max value for data type
* wave: like the dummy cameras in cockpit
* beam: bright beam somewhere on the image (which is what I wold like now to experiment with BeamDelta)

mickp, 2019-03-07:
Agreed. This has been on my list for some time. I recently modified the TestCamera (not yet merged) to add random variations to the limits used for random noise so we can test rescaling in the client properly, and I very nearly made this depend on programmable parameters, but decided against it.

I think we could have different modes, and expose them via the settings framework as an Enum setting, so they appear as a dropdown in the cockpit settings interface. If any modes need parameters, we can create defaults in the settings dict when we enable that mode. I think the cockpit settings UI will tolerate this, as it'll just create controls for the new settings when it finds them ... although it wouldn't yet clear up unused settings.

mickp, 2019-05-17:
Added in 6ad303f48f0f04c6ca14a4fc21a8190743721c5f

closed 2019-05-17 by mickp