python-microscope/microscope #74: throlabs filterwheel does not work

carandraug, 2019-04-18:
I haven't got around to actual try it, but looking at the code, I'm pretty confident that it does not work, because the com port is passed as list of filters.  Basically, the code is this:

```
class FilterWheelBase(Device):
    def __init__(self, filters, *args, **kwargs):
        super().__init__(*args, **kwargs)
        ## rest of code assumes filters to be a list of tuples

class ThorlabsFilterWheel(FilterWheelBase):
    def __init__(self, com, baud, timeout, **kwargs):
        ## com port is passed as first argument to the parent constructor
        ## which expects it to be a list of filters
        super().__init__(com, baud, timeout, **kwargs)
````

I have looked back in history and seems like it has always been this way. Are we actually using this anywhere?

mickp, 2019-04-18:
I thought we were running it at Diamond ... but perhaps not.

mickp, 2019-05-08:
Fixed by #80

closed 2019-05-08 by mickp