python-microscope/microscope #81: Setting trigger_mode to 'External Exposure' throws error

juliomateoslangerak, 2019-05-08:
With the Zyla, if I set the trigger_mode to 'External Exposure' I get this error when applying.  Sorry for the frenchified windows error message (the specified module was not found). All the other trigger_modes do not throw errors.

Traceback (most recent call last):
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\gui\device.py", line 338, in <lambda>
    applyButton.Bind(wx.EVT_BUTTON, lambda evt: self.device.updateSettings(self.current))
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\interfaces\imager.py", line 79, in wrapper
    result = func(*args, **kwargs)
  File "C:\Users\julio\PycharmProjects\cockpit\cockpit\devices\microscopeCamera.py", line 113, in updateSettings
    self.proxy.update_settings(settings)
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\site-packages\pyro4-4.73-py3.6.egg\Pyro4\core.py", line 185, in __call__
    return self.__send(self.__name, args, kwargs)
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\site-packages\pyro4-4.73-py3.6.egg\Pyro4\core.py", line 467, in _pyroInvoke
    data = serializer.deserializeData(msg.data, compressed=msg.flags & message.FLAGS_COMPRESSED)
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\site-packages\pyro4-4.73-py3.6.egg\Pyro4\util.py", line 170, in deserializeData
    return self.loads(data)
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\site-packages\pyro4-4.73-py3.6.egg\Pyro4\util.py", line 451, in loads
    return pickle.loads(data)
  File "c:\users\julio\pycharmprojects\microscope\microscope\cameras\SDK3.py", line 37, in <module>
    _stdcall_libraries['ATCORE'] = ctypes.WinDLL('atcore')
  File "C:\Users\julio\Applications\WinPython\python-3.6.5.amd64\lib\ctypes\__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] Le module spécifié est introuvable

juliomateoslangerak, 2019-05-08:
Sorry I correct, the error is when setting to 'External Start'

mickp, 2019-05-08:
It looks like the client is trying to import the SDK3 module, presumably to
deserialise an object defined in that module. If the CDLL/WinDLL reference
is created at the root of the module (i.e. not in a class or function) and
the DLL is not available on that machine, then python will throw a module
not found exception on import. The SDK3 stuff probably needs wrapping in
the same way as those modules in the _wrappers submodule to prevent this.
In the mean time, just put the DLL somewhere python can find it on the
client.

juliomateoslangerak, 2019-05-08:
If I set the trigger mode to 'software' I get an external trigger. Either external or external exposure. It is not external start (I get a single image) and it is not software (disconnecting the trigger results in no images). Interestingly, in andorSDK initialize method there is a default configuration setting the exposure_mode to 'software' 
`        self._trigger_mode.set_string('Software')`

It seems to me that the trigger modes and indexes are a bit messed up and defined in a few different places...

juliomateoslangerak, 2019-05-13:
I did some tests and here is what I get:
```python
import Pyro4
cam = Pyro4.Proxy('PYRO:AndorSDK3@10.6.19.31:8001')
cam.enable()
Out[4]: True
cam.disable()  # I disable here but if I keep the camera enabled I get similar results
cam.get_setting('trigger_mode')
Out[6]: 4
cam.describe_setting('trigger_mode')
Out[7]: 
{'cached': False,
 'readonly': False,
 'type': 'enum',
 'values': [(0, 'Internal'),
  (1, 'External Start'),
  (2, 'External Exposure'),
  (3, 'Software'),
  (4, 'External')]}
cam.set_setting('trigger_mode', 3)  # Let's set the trigger mode to software
cam.get_setting('trigger_mode')
Out[9]: 3
cam.get_setting('exposure_time')
Out[10]: 0.0098304
cam.describe_setting('exposure_time')  # In software trigger mode we should be able to change the exposure time...
Out[11]: 
{'cached': False,
 'readonly': True,
 'type': 'float',
 'values': (0.0009984, 30.0)}
cam.set_setting('exposure_time', 0.2)
cam.get_setting('exposure_time')  # But we cannot. It's really read-only
Out[13]: 0.0098304
cam.set_setting('trigger_mode', 2)  # So now to external exposure
cam.describe_setting('exposure_time')
Out[15]: 
{'cached': False,
 'readonly': False,
 'type': 'float',
 'values': (9.6e-06, 30.0)}
cam.set_setting('exposure_time', 0.2)  # I guess this is might be ok as we need to change exposure time to get the total readout time
cam.get_setting('exposure_time')
Out[17]: 0.19999679999999997
```

juliomateoslangerak, 2019-05-13:
The SDK seems to give me the wrong indexes

juliomateoslangerak, 2021-06-02:
This is working now.

closed 2021-06-02 by juliomateoslangerak