python-microscope/microscope #91: deviceserver will not work with classes needing host, port, or cls argument

carandraug, 2019-05-31:
The deviceserver takes a dictionary with a device definition. This dictionary has the host and port to serve the device, the device class, and all other items are arguments to the class constructor. This limits the arguments that can be passed to the device since their names must not clash with the other keys.

This simplest solution is to separate the arguments for the device from the arguments to the device server. So instead of usining something like this:

```
{
  'cls' : testdevices.TestLaser,
  'host' : '127.0.0.1',
  'port' : 8001,
  'com' : 6,
  'baud' : 115200,
}
```

to use this:

```
{
  'cls' : testdevices.TestLaser,
  'host' : '127.0.0.1',
  'port' : 8001,
  'args' : [],
  'kwargs' : {
    'com' : 6,
    'baud' : 115200,
  },
}
```

We could also not include `args` since I never saw we using it (and all the cameras would discard them anyway). This makes a new modified `device` function simpler too.

carandraug, 2019-05-31:
Proposed solution in 89aef4a (or device-def-kwargs-issue91 branch)

carandraug, 2019-06-06:
I would like to have this reviewed sooner than later. The change is small and does not make the deviceserver configuration more complicated. However, it is not backwards compatible, which is why I'd rather address it sooner because later there may be more systems running it.

Basically, with this change, device server configuration looks like this (commented lines is how they look like now):

```
DEVICES = [
#    device(SaphireLaser, '127.0.0.1', 8000,
#           com='/dev/ttyUSB0', baud=19200),
    device(SaphireLaser, '127.0.0.1', 8000,
           dict(com='/dev/ttyUSB0', baud=19200),

#    device(IDSuEye, '127.0.0.1', 8001,
#           serial_number='4103350857'),
    device(IDSuEye, '127.0.0.1', 8001,
           {'serial_number' : '4103350857'}),

    ## The uid device server argument changed place so it either
    ## appears after kwargs or must be named
#    device(atmcd.AndorAtmcd, '127.0.0.1', 8000, 'VSC-01234'),
    device(atmcd.AndorAtmcd, '127.0.0.1', 8000, uid='VSC-01234'),
]
```

The change to the uid argument to the device server is that `uid` is less likely to be needed (it is only needed for floating devices, while pretty much all devices will need `kwargs`).

mickp, 2019-06-06:
Looks OK, although I think using ```kwargs``` as the name might cause some confusion (I had to reread a couple of times to be certain that kwargs was referring to the dict, and not what ```kwargs``` usually refers to). Perhaps it should be ```devargs```, or something?

carandraug, 2019-06-06:
Yeah, that's fair. I thought the same when writing it. I don't like `devargs` because `args` is meant for list of arguments and we have keywords of arguments. So maybe `dev_kwargs`. Or `init_kwargs` which is what I had in a previous iteration of this commit and makes it clear these are the kwargs used to call `__init__`?

mickp, 2019-06-06:
Or maybe just ```conf```?

carandraug, 2019-06-07:
I'm cool with `conf`. I will amend the commit and push then.

closed 2019-06-07 by carandraug