python-microscope/microscope #94: Thorlabs filter wheel doesnt start.

iandobbie, 2019-06-11:
Trying to connect to a thorlabs filter wheel returns....

object has no attribute '_logger'

Is this a python3 thing?

carandraug, 2019-06-11:
I'm guessing what is happening is that it's failing to construct the device, and I guess this is happening because it's using some invalid argument (or I somehow messed it up while fixing #84).

The error message is weird but that's a separate bug (see one of the comments in issue #69). Basically what happens is that when `__init__` fails, `__del__` is called which then calls `shutdown` which tries to use log being shut down. However, if `__init__` errors before you create the logger, then `shutdown` will error again.

carandraug, 2019-06-11:
Can you show how you were constructing or connecting to device?

iandobbie, 2019-06-12:
Ok. It is probably missing vital setup info. I was just trying as basic a startup as possible. The tests I was doing are on CryoSIM on the laser control machine 192.168.0.10 in c:/microscope_config/laser_config.py.

I will look at diamond to see what the filter wheel have there in their config.

Ian

NickHallONBI, 2019-06-12:
For reference, the error is:

```
>>> thorlabs.ThorlabsFilterWheel(com='com10', baud=9600)
Exception ignored in: <function Device.__del__ at 0x000002B3F80127B8>
Traceback (most recent call last):
  File "c:\microscope\microscope\devices.py", line 207, in __del__
    self.shutdown()
  File "c:\microscope\microscope\devices.py", line 255, in shutdown
    self._logger.info("Shutting down ... ... ...")
AttributeError: 'ThorlabsFilterWheel' object has no attribute '_logger'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: __init__() missing 1 required positional argument: 'timeout'
```

carandraug, 2019-06-12:
The problem was that the `timeout` argument (which is required) is not being specified. Not an issue in microscope.

However, for other devices we have a default timeout and baudrate. The baudrate is on the filterwheel manual. Do we have a sensible default for timeout?

mickp, 2019-06-12:
A default serial timeout of a few seconds should work. The serial
connection becomes unresponsive during wheel moves, and moves can take a
few seconds.

carandraug, 2019-06-12:
For wheel moves, the manual says typical values are:

* FW102C (adjacent location) : < 1 sec
* FW212C (adjacent location) : < 0.5 sec
* FW102C and FW212C (other locations): 2.5 sec

We could use a lower timeout for all other communications and change it as required for wheel moves. What do you think?

carandraug, 2019-06-12:
I have checked the code and for wheel moves the code already waits up to 10 times before raising an exception.

I have set the default baudrate to 115200 (what is on the manual) and the timeout to 2 seconds.

closed 2019-06-12 by carandraug