python-microscope/microscope #96: add support for ASI stages

carandraug, 2019-07-09:
Support for ASI stages was requested at MMC 2019.

jondaniels, 2022-09-30:
ASI is would consider providing and supporting device-specific code for this project just like we do for Micro-Manager.  Or alternatively if a project expert wants to tackle the task, here is the [ASI command documentation](http://asiimaging.com/docs/command_quick_start).

ASI offers stages, filter wheels, analog output devices, and digital triggering devices.  So I assume we would use the "controller" device?  

I found the page [Support for New Device ](https://www.micron.ox.ac.uk/software/microscope/doc/get-involved/new-device.html), but if there are other places to go for more information please share them.

Tagging @bls337 as the most likely contributor.

iandobbie, 2022-09-30:
This sounds like a very interesting proposal. We are certainly keen on having as wide a hardware support base as possible. I don't currently have any ASI hardware so would really struggle to implement any hardware control.

On the more specific question. I think that the ASI setup will be a controller which will export devices like a stage, filter wheel, shutter, etc.....

juliomateoslangerak, 2022-10-07:
Indeed very interesting. 
I have an ASI 1 axis z drive (LS-50 I believe) that I can test in cockpit.
We also have an MS2000 controller with a XY stage on a different machine that is not using microscope but I could run tests on it.

iandobbie, 2023-03-03:
Turns out my 3i system actually has an ASI controller and we might well be scavenging it as it has a casdcade II and no way to talk to it from windows 10. I will have a look at starting to write support as this seems like a no brainer.

iandobbie, 2023-03-17:
Just spend about 2 hours trying to talk to the stage and getting nowhere. Gave up, went back to my desk and noticed that the documentation said it was connected with a null-modem cable.  Just plugged into the other end of the cable from the computer rather than directly into the controller and it worked. What as waste of an afternoon. 

Remember when you had to properly sort out Rx Tx connections? when ethernet patch cables were wired differently from the one going to your computer. The joys of auto- negotiation, but also how quickly we forget. I'm sure a decade or two ago I would have thought to try this hours ago.

iandobbie, 2023-04-20:
This is starting to work with a few edits from the ludl controller for line endings and ensuring moves finish before we ask for another move. However although it appears to work the y axis is failing. It appears to move to the limits fine, but then after finding them an moving to the central position, it doesn't make it to the central position and then will not move any more. Very strange as the identical code seems to work fine for X.

I'll keep working on it.

jondaniels, 2023-04-20:
I'm happy to glance at the code in case I can spot something, but where is it?

iandobbie, 2023-04-20:
Its in my fork at https://github.com/iandobbie/microscope/tree/ASIstage
I am testing this using my laptop with the asi stage as a real device but then a bunch of simulated devices and cockpit to do the actual control. 
The code is a bit complex as the control approach has 2 axis, making up the stage, and then above that a controller which could also allow access to things like filter wheels etc... Currently only the xy stage is implemented.

jondaniels, 2023-04-20:
I can't see immediately amiss at first glance.  If you send a serial log that could help.  It would also be good to know the firmware version and compile date (accessible with "CD" and "V" commands).

I don't think it matters much but you are using the long versions of the serial command names.  If you like you can use shorter ones which will make things slightly faster.
- I for INFO
- / for STATUS
- R for MOVREL
- M for MOVE
- @ for SPIN
- RS for RDSTAT
- S for SPEED
- W for WHERE

You can find documentation for each of these commands using the long name with base URL like [http://asiimaging.com/docs/commands/RDSTAT](http://asiimaging.com/docs/commands/RDSTAT) and so forth.

iandobbie, 2023-04-21:
Thanks for that suggestion. I left the long versions to make it more understandable, but with the relatively slow link (9600) the transfer time is not insignificant. I could also change the baudrate to reduce IO time.

iandobbie, 2023-09-20:
@joris-afk from McGill is trying to interface to an ASI stage. I have some code that is at least partially working in my fork mentioned above  https://github.com/iandobbie/microscope/tree/ASIstage. This code fails to properly initialize my stage. The X axis is fine the Y stalls, which might be a problem with my stage. I have not yet got to the relevant trouble shooting, but can try here relatively easily.

jondaniels, 2023-09-20:
If you post a log of the serial traffic to/from the controller I am happy to look it over for anything suspicious.

iandobbie, 2023-09-20:
Hi Jon, Sorry I know i promised to previously and push it down the todo stack. I will have another go soon. I will try tomorrow morning but might be stuck at home as one of my kids is unwell.

iandobbie, 2023-09-21:
I am not sure what is happening. Here is a record of some of the conversation:

In the attached file I have removed repeated polls for position etc which just return the same value: 
Lines that start "send %s"  are messages sent to the controller (even my debug code has bugs!)
Lines starting "read line" come back from the controller, other lines are other piece of debug info from my code.

1) I connect and get the controller info
2) Set speed on x and y
3) calibrate x and y by spin -ve to stop, set to zero, spin +ve till stop read value, move to 1/2 way point on axis. 

4) move x to 483019 - this works
5) move y to 542335 
This starts returning "A 56"
Then my code seems to think its finished and says "got to  541120.0"

Expanding the response (56 = 0x111000) implies that the limit switch is stuck maybe?

iandobbie, 2023-09-21:
Now with attachment!
[asi-debug.txt](https://github.com/python-microscope/microscope/files/12689895/asi-debug.txt)

iandobbie, 2023-09-21:
One further data point. 

If I redo the sequence the Y axis doesn't move at all, even in the spin to min and spin to max sections which produces a avaliable area of 0 to 0 so then nothing works. 

However if I power cycle the stage controller and do it again I get the result as above it does the homing (scan to min set to zero, scan to max, move to mid point) then wont move again.

jondaniels, 2023-09-21:
Setting the speed to 10000 will make it the maximum possible.  Speed is set in units of mm/s, not the axis units which is by default 0.1um per unit.  ([SPEED command documentation](http://www.asiimaging.com/docs/commands/speed))

I'm guessing that the Y axis is disabling itself because it is falling behind the planned trajectory when you request the maximum allowable speed.  (Due to stage construction, one axis is moving the middle plate in addition to the top plate and sample, and depending on leadscrew you may not be able to attain in practice the maximum settable speed when loaded).  If this happens you will see on the LCD screen "D" next to the axis when this happens, and then you can use the [MOTCTRL command](http://www.asiimaging.com/docs/commands/motctrl) to re-enable the axis.

A couple other notes: 

Spin is also limited to +/- 128 but the firmware should cap the value internally.  Spin is a very rudimentary command and not generally used.  ([SPIN documentation](http://www.asiimaging.com/docs/commands/spin))  If you are trying to run the stage to the magnetic limit switch, the easiest way is to make a move to a very large or very small position.  I don't think there is necessarily a problem with your implementation using SPIN instead of a very large move, but it is unconventional.

To poll the move status of a particular axis more easily you can use `RDSTAT <axis>?` -- note the "?" at the end -- which will return either "B" or "N" after the ":A" which is easier to parse than the decimal code where you are fishing out a particular bit.  ([RDSTAT documentation](http://www.asiimaging.com/docs/commands/rdstat)).

iandobbie, 2023-09-21:
I see I have some serious documentation to read. 

My spin approach was I had no idea about the units and didn't know what the stage would do if I tried to move somewhere it couldn't get to. I can easily recode it as move to somewhere very -ve and very +ve. 

I will try to slow it down a lot and see what happens, thanks.

iandobbie, 2023-09-21:
Thanks for the pointers above. I changed the spin to +/- 100 and the speed to 5 mm/s and now everything is working properly. I will have to investigate what are acceptable speeds for x and y but at least it is working now.

iandobbie, 2023-09-21:
Merged support into master branch with commit 0c89fdc4acf1715261c97fa2f5250352f93bdc31. Marking as closed will open new issue with updates needed

closed 2023-09-21 by iandobbie

jondaniels, 2023-09-21:
Max allowable speed depends on leadscrew pitch of the stage.  The approach Micro-Manager takes is to set an outrageously high speed (e.g. 1000), then query the speed (which the firmware caps at the max settable speed), and then back off from there as desired.

iandobbie, 2023-09-22:
Moved the discussion to issue #283.