python-microscope/microscope #98: support for AFM devices

carandraug, 2019-07-09:
At MMC 2019, it was requested that we include devices for AFM, or at least think about them when designing the interface.

mickp, 2019-07-09:
Any idea what aspect of the AFM they wanted to control? They generally have
proprietary controllers which are not particularly open, and vary hugely
between manufacturers.

carandraug, 2019-07-09:
Whoever wrote down on the poster just wrote "AFM client (anything SPM related)". There was another person that did spoke to me about it and seems like they were building their own. I have no clues about AFM but she wanted to control a laser and a camera. She sold it to me as a laser reflecting on the tip of a needle and then acquiring the position of the reflected laser on a camera. No mention of specific vendors though.

mickp, 2019-07-09:
Yep - you measure the deflection of a cantilever by detecting the position
of a reflected spot. You're not going to do that fast enough with any of
the cameras we support, though - you need a few MHz bandwidth, so usually
use a (2- or 4-) segmented photodetector with transimpedance amplifiers to
measure voltage differences between the segments. These are fed into a
lock-in amplifier or PLL to drive the Z-position in a servo loop on
deflection, oscillation amplitude or frequency.

* We can support stages for home-built AFMs - they'll be similar to the
higher-end stages used for light microscopy.
* We can support cameras for sample overview.
* We may be able to support some amplifiers by toggling servo activity, and
perhaps modifying set points and gains.