python-microscope/microscope #101: andorsdk3: zyla camera fails to initialise with missing _trigger_mode attribute

carandraug, 2019-08-06:
Reported by @iandobbie by email:

> The Zyla camera fails to initialise at line 368 of the andorsdk3.py
> module as the object doesn't have an attribute _trigger_mode. Looking at
> the code with David it appears as though the camera doesn't have this
> feature and the initialise routine is therefore deleting the it. However
> the camera is a Zyla and must have trigger modes (I assume?). We need to
> investigate further. Is it a change in the andor SDK between versions?
> or maybe the USB version just doesn't have this parameter?

mickp, 2019-08-16:
The SDK cares about case in the feature names. I don't know how or when this happened: I saw the same error with the existing SDK installation on DeepSIM, and Ian and I are both pretty sure we had the Zyla running on there a while back. I added a dict to map AndorSDK3 class attributes to SDK3 feature names in 987d3720c534a28e7994c7415a01706c13e8a883, now merged into master. 

I can now initialize and query features on SIMCAM (the SDK's software fallback). I see an AT_CONNECTION_ERROR with the real camera, but this looks like a hardware or SDK state problem rather than a microscope problem. @iandobbie, @toshikikubo - can you please test on the system in engineering and report back?

mickp, 2019-08-16:
OK. So version 3.8.30007:
* will not connect to the CameraLink Zyla on DeepSIM;
* and won't enumerate (i.e. doesn't even see) the USB Zyla from engineering, depite USB support being added in 3.8.30005.

Meanwhile, Julio's zyla appears to be working fine, even when using all-lowercase feature names. 

Working theory: connecting our cameras to a newer version of Solis has updated their firmware, and they now have conflicts with the SDK version we have available. 

I'll take a look at what's in their current driver pack they distribute for setting up MicroManager.

mickp, 2019-08-23:
The dict of feature names seems to have fixed the problem.
No Zyla works on my development machine, for some reason. The USB Zyla works fine on another machine.

closed 2019-08-23 by mickp