python-microscope/microscope #112: Optimos sCMOS camera support

will-h-uon, 2019-10-22:
Optimos sCMOS camera is connected by a proprietary PCIe card. Their documentation says it does not support python - does this mean I'm at a dead end here?

carandraug, 2019-10-22:
Almost no camera vendor supports Python. They all seem to have a C library so you have to wrap it with ctypes and then make use of it. That's how current cameras do it. The deformable mirrors are the same but much simpler.

For example:
* [BMC deformable mirror module]https://github.com/MicronOxford/microscope/blob/master/microscope/mirror/bmc.py
* [libbmc wrapper](https://github.com/MicronOxford/microscope/blob/master/microscope/_wrappers/BMC.py) (the BMC C library)

iandobbie, 2019-10-22:
The optimos appears to be a qimaging camera and so should be supported under pvcam. It is however a legacy device so I don't know how recent a pvcam it is compatible with and if it will work out of the box. The relevant web page (https://www.qimaging.com/software) suggests it is supported by the latest pvcam so it might just work. 

Try installing pvcam then you need to install microscope (https://github.com/MicronOxford/microscope) and define a camera like 

import microscope.cameras.pvcam as pvcam
DEVICES=[device(pvcam.PVCamera, '127.0.0.1', 8000)

then in  the cockpit depot.conf you need an entry like

[pvcamera]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
uri: PYRO:PVCamera@127.0.0.1:8000

then it should hopefully work. Good luck!

will-h-uon, 2019-10-23:
Thanks - pvcam does support the sCMOS out of the box.

Regarding depot.conf, I'm unable to find a copy of the file on my computer - where should it be?

This message and any attachment are intended solely for the addressee
and may contain confidential information. If you have received this
message in error, please contact the sender and delete the email and
attachment. 

Any views or opinions expressed by the author of this email do not
necessarily reflect the views of the University of Nottingham. Email
communications with the University of Nottingham may be monitored 
where permitted by law.

carandraug, 2019-10-23:
> Regarding depot.conf, I'm unable to find a copy of the file on my computer - where should it be?

The `depot.conf` file is the configuration for cockpit only. The default location of this file will be system dependent. See the [cockpit manual](https://www.micron.ox.ac.uk/software/cockpit/config.html#depot-configuration) but if you are on windows, this could be `C:\ProgramData\cockpit\depot.conf` . The file probably does not exist, you will need to create it.

will-h-uon, 2019-10-24:
Thanks for that. Unfortunately I'm still having difficulties: When I run the following
from microscope.devices import device
import microscope.cameras.pvcam as pvcam
camera = pvcam.PVCamera();
camera.enable()
I get an error:
Traceback (most recent call last):
  File "Will\junk\testcam.py", line 4, in <module>
    camera.enable()
  File "c:\users\labuser\pip\microscope\devices.py", line 449, in enable
    raise err
  File "c:\users\labuser\pip\microscope\devices.py", line 445, in enable
    result = self._on_enable()
  File "c:\users\labuser\pip\microscope\cameras\pvcam.py", line 1252, in _on_ena
ble
    self._params[PARAM_EXP_RES].set_value(EXP_RES_ONE_MILLISEC)
KeyError: 151191554
And if I try to use remote devices, as suggested by Ian, using this code
import microscope.clients
camera_uri = 'PYRO:PVCamera@127.0.0.1:8000'
camera = microscope.clients.DataClient(camera_uri)
camera.enable()
I get this error:
Traceback (most recent call last):
  File "C:\Users\LabUser\AppData\Local\Programs\Python\Python37-32\lib\site-pack
ages\pyro4-4.77-py3.7.egg\Pyro4\core.py", line 515, in connect_and_handshake
    sslContext=sslContext)
  File "C:\Users\LabUser\AppData\Local\Programs\Python\Python37-32\lib\site-pack
ages\pyro4-4.77-py3.7.egg\Pyro4\socketutil.py", line 307, in createSocket
    sock.connect(connect)
ConnectionRefusedError: [WinError 10061] No connection could be made because the
 target machine actively refused it

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "Will\junk\testclientcam.py", line 3, in <module>
    camera = microscope.clients.DataClient(camera_uri)
  File "c:\users\labuser\pip\microscope\clients.py", line 62, in __init__
    super().__init__(url)
  File "c:\users\labuser\pip\microscope\clients.py", line 41, in __init__
    self._connect()
  File "c:\users\labuser\pip\microscope\clients.py", line 46, in _connect
    self._proxy._pyroGetMetadata()
  File "C:\Users\LabUser\AppData\Local\Programs\Python\Python37-32\lib\site-pack
ages\pyro4-4.77-py3.7.egg\Pyro4\core.py", line 615, in _pyroGetMetadata
    self.__pyroCreateConnection()
  File "C:\Users\LabUser\AppData\Local\Programs\Python\Python37-32\lib\site-pack
ages\pyro4-4.77-py3.7.egg\Pyro4\core.py", line 596, in __pyroCreateConnection
    connect_and_handshake(conn)
  File "C:\Users\LabUser\AppData\Local\Programs\Python\Python37-32\lib\site-pack
ages\pyro4-4.77-py3.7.egg\Pyro4\core.py", line 549, in connect_and_handshake
    raise ce
Pyro4.errors.CommunicationError: cannot connect to ('127.0.0.1', 8000): [WinErro
r 10061] No connection could be made because the target machine actively refused
 it

It looks to me like there's some problem with the configuration of Pyro, although that's just a guess, and I have no idea why using camera.enable() causes an error, even though camera.set_exposure_time() does not.

mickp, 2019-10-24:
That looks more like a missing key error in self._params. It could be that
this camera doesn't set exposure time in the way the hardware I had
available
when I wrote this module. We should probably test that the key is in the
dict near lines 1249 and 1252 of pvcam.py, and omit setting the exposure
resolution if the key is missing.

iandobbie, 2019-10-29:
I suspect that Mick is right as he wrote the pvcam module. I will investigate getting hold of an optimos to test with.