python-microscope/microscope #113: Aurox Clarity device issues.

iandobbie, 2019-10-29:
David's latest updates have broken the Clarity device. On startup it generates an error:

TypeError: Can't instantiate abstract class Clarity with abstract methods _on_shutdown, initialize

carandraug, 2019-10-30:
The mentioned changes that broke the Clarity is commit 83f1256cdfa1 . Seems like our ABCs never worked in Python 3. That change fixed it so all devices that didn't implement the Device abstract methods are now failing like they should. I went now through the devices and seems to me that the LinkamStage also does not implement the `initialize` and `_on_shutdown` methods.

carandraug, 2019-10-30:
I have now fixed the issue for the LinkamStage in c0a70733301f . I went through the other classes and seems like they all have the required methods. Closing as fixed.

closed 2019-10-30 by carandraug