python-microscope/microscope #116: Improve complicated startup process.

iandobbie, 2019-11-06:
On CryoSIM yesterday it was pretty clear we need a better startup  process. I haven't been on the system in a while and needed to work out what was where and start individual deviceservers. 

Could we knock up a basic ssh script which would do 

ssh host1 screen -d -m ' deviceserver config1.py' 
ssh host2 screen -d -m  'deviceserver config2.py'
etc....

then we could have it as a batch file, or shell script on the master computer.

mickp, 2019-11-06:
Possibly, but Windows doesn't come with an ssh server, so we'd need to install one and figure out what permissions to give it. (This is why I wrapped the old device remotes in a Windows service.)

carandraug, 2019-11-06:
Are you suggesting this as a solution for CryoSIM only? Or are you suggesting we create something general enough for all cases of microscope users?

iandobbie, 2019-11-07:
It was pretty obvious that the startup from scratch process is hard and it would be nice to have a mechanism that could start all the remote deviceservers from one command. This would be useful on Cryosim, but also any other system which requires servers on multiple computers. In the mean time the solution is I guess rdp and batch files on the desktops.

carandraug, 2019-11-07:
To start all the servers, there's systemd and sysinitv on linux and windows services on windows. deviceserver works with all of those. I guess we should write documentation for that though, even though that's not technically microscope documentation.

juliomateoslangerak, 2019-12-20:
Our Abberior STED uses a Ethernet remotely controlled power supply. **It is a beauty!!** 
- You start the software
- Software prompts if you want to initialize hardware. You might imagine that here you might choose between different HW configs, or you might just choose which specific devices you are going to use (typically lasers)
- You click start and the required devices are powered through that remote power supply.
- Then you might start the different device servers

Upon quiting the software is asking if you also want to shut down the hardware.
If you star again devices that are on are showing that they are on.

juliomateoslangerak, 2019-12-20:
I believe it is something like this:
https://www.apc.com/shop/us/en/products/Rack-PDU-Switched-2U-30A-208V-16-C13/P-AP7911B

iandobbie, 2023-03-03:
Great idea but slightly different from the original issue. I think we should close this as the rdp solution kind of works and I cant think of any other cross platform solution. Maybe future windows system for linux will give us SSH but I dont hold out much hope

closed 2023-03-03 by iandobbie