python-microscope/microscope #117: Linkam stage is missing pyro expose

iandobbie, 2019-11-07:
When we connect from cockpit (updated to lastest master 7/11/2019) with linkam stage on microscope (also latest master) we get...

 File "C:\Program Files\Python37\lib\site-packages\Pyro4\core.py", line 280, in __getattr__
    raise AttributeError("remote object '%s' has no exposed attribute or method '%s'" % (self._pyroUri, name))
AttributeError: remote object 'PYRO:LinkamBase@192.168.0.10:8001' has no exposed attribute or method 'get_position'

carandraug, 2019-11-07:
The lack of `Pyro4.expose` should not make a difference because deviceserver runs with `REQUIRE_EXPOSE` set to False.

carandraug, 2019-11-07:
I see the problem. The deviceserver is serving an instance of `LinkamBase` instead of `LinkamCMS`. Maybe we should mark `LinkamBase` class as private?

iandobbie, 2019-11-07:
Maybe so but the issue boils down to user error, maybe due to inadequate documentation, user training, or just the fact I didn't read any documentation which does exist? I think as it was a config error we should close the issue.

closed 2019-11-07 by iandobbie

carandraug, 2019-11-07:
On the client side, Pyro jst says that the method is not being exposed. The client does not know whether it is because of a missing exposed or not existing at all on the server side. Maybe we can have cockpit check for the presence of all abstract methods which would give a nicer earlier error.

But if `LinkamBase` is only meant to be used inside the linkam module, marking it as private will help prevent this specific issue.

But yeah, for the most complex modules, it's not always clear which of the classes is meant to be served and the lack of documentation just makes it worse.

mickp, 2019-11-08:
Base classes made private in d9cb040f896543a47e84a1ef588fa78415c6bacb .