python-microscope/microscope #119: filterwheel: not clear whether filter positions are counted from 0 or 1

carandraug, 2019-11-21:
It's not documented if the filter positions in a filterwheel device are counted from 0 or from 1. The base class sets the valid range of positions between 0 and the number of positions, and zero is the default answer from `get_position`. Reading the documentation from thorlabs, it seems that at least the thorlabs implementation counts them from 1.

carandraug, 2019-11-25:
I just spoke with @mickp about it and he mentions that not all filterwheels are numbered. Some may have filter A, B, C, D, and sometimes even words like CAL for calibration.

We can't have some concrete classes taking integers and others taking strings so we settle on strings.

carandraug, 2020-07-19:
We spoke about this again, and decided that we don't want to have to deal with strings for this. The vast majority of filterwheels uses numbers and when we find those who don't, then we'll just document how those map from their names to numbers.

carandraug, 2021-01-07:
As far as I can see, we need to check what really happens on the following:

- [ ] Thorlabs
- [ ] Aurox
- [ ] Prior

mickp, 2021-01-07:
How about using an Enum which maps 0-based integers to whatever the filter wheel uses?

carandraug, 2021-01-08:
An enum like that seems like a good idea. The client could then use it to provide something readable to the user. This will be device specific, and implemented for that device when a device comes that does not use integers.

I have now checked on Thorlabs (with the FW102C which is on the biochemistry cryosim) and fixed it (it was using base 1). I don't remember what prior did and their documentation does not say (and we don't have a prior filter wheel anymore, it was borrowed from Alan Tilly).