python-microscope/microscope #120: add support for zaber devices

carandraug, 2019-11-21:
We are now playing with [zaber devices](https://www.zaber.com/products).  They seem to talk RS232, the [manual is online](https://www.zaber.com/wiki/Manuals/ASCII_Protocol_Manual), and they have a [console program to control zaber devices](https://www.zaber.com/wiki/Software/Zaber_Console) which is open source so we can inspect how they do some things.

Their devices can be [daisy chained](https://www.zaber.com/products/optical-mounts/X-FWR-E/easy-set-up) and will all share the same port so to avoid tangling messages from different devices, they can be better modelled as controller devices.

carandraug, 2019-11-22:
I have added support for Zaber filter wheels, filter cube turrents, and stages on my "wip-zaber" branch. The stage interfaces uses the same I was experimented with (and not yet merged) for the prior stages.

carandraug, 2019-11-25:
If no has any issue with at least 63365ead8862ca6de23b68b57532b575414aee05 (from the branch mentioned above) I will push it which adds supports for Zaber filterwheels.

iandobbie, 2020-10-09:
We now have a purchased system with a few outstanding issues.

- [ ] Updates in cockpit are 1 position behind the current position.

- [ ] Filter wheel moves produce errors in cockpit.

The second issue is that the position is interrogated while the wheel is still moving producing a result os position 0 which is undefined with the zaber filter wheel. waiting for the position call to retunr a non zero result eg: 
'        while(self._conn.get_current_index(axis=1)==0):
            time.sleep(0.1)
        return self._conn.get_current_index(axis=1)'
Prevents the error on cockpit

iandobbie, 2020-10-09:
I have the start of a working LED module added in https://github.com/iandobbie/microscope/tree/wip-zaber-IMD. Connects to the LED module, added individual LEDs as new devices to the controller and does enable/disable and staus calls. The power setting doesn't work and I'm not sure quite why.

iandobbie, 2020-10-12:
Worked out the power issue. Now seems to be basically fully working in my branch with the following deviceserver setup file
[microscope-zaber.txt](https://github.com/MicronOxford/microscope/files/5366105/microscope-zaber.txt)

iandobbie, 2020-10-12:
Then I connect to it from cockpit with 
[depot.conf.txt](https://github.com/MicronOxford/microscope/files/5366110/depot.conf.txt)

iandobbie, 2020-10-14:
Just realised that thew joystick is now misconfigured. I renumbered the devices as two of them clashed and the joystick now tries to talk tot he wrong device. The manual suggests using their windows only tool to reprogram the joystick, not an easy option for me. Maybe we need some code to wrap the joystick programming functions?

carandraug, 2020-10-20:
I now pushed my previous changes, with Ian's changes, and more fixes on top. This adds support for zaber stages (xy and z), LED controllers, filter wheels, and cube turrets. The depot configuration mentioned above will not work anymore since, instead of having the LEDs exposed on the daisy chain itself, there is only the LED controller (the LEDs need to be accessed via the LED controller). So something like this for cockpit:

```
[zaber chain]
type: cockpit.devices.microscopeDevice.MicroscopeBase
uri: PYRO:ZaberDaisyChain@localhost:8000

[led controller]
type: cockpit.devices.microscopeDevice.MicroscopeBase
controller: zaber chain
controller.name: 3

[LED2]
type: cockpit.devices.microscopeDevice.MicroscopeLaser
controller: led controller
controller.name: LED2
wavelength: 473

[LED3]
type: cockpit.devices.microscopeDevice.MicroscopeLaser
controller: led controller
controller.name: LED3
wavelength: 568
```

Anyway, closing this as fixed.

closed 2020-10-20 by carandraug