python-microscope/microscope #121: Add support for clarity processing

mickp, 2019-11-27:
After playing around with creating dynamic classes and monkeypatching them, I think the way forward is to add a processing pipeline to DataDevice. The device will do it's own processing first, then push the data through each stage in the pipeline before returning. 
In the first instance, the pipeline can just be a list of functions that take one argument, data, and return that argument. Any object that wants to add to the pipeline can do so by adding a function to the list, using partials where that function needs to reference the object itself. 
e.g. If ```SomeObject``` does processing in a method ```process_data(self, data) -> data```, using its own attributes as processing parameters, it can add it to the pipeline using ```SomeDataDevice.pipeline.add(functools.partial(self.process_data, self)```

mickp, 2019-11-27:
[Work in progress](/mickp/microscope/tree/121-pipeline)

iandobbie, 2020-11-06:
Danny worked on integrating this properly on the engineering system. His current version is at https://github.com/dstoychev/microscope/tree/aurox

There is however an outstanding issue:

> There is still the issue of the clarity_processor modules
> (https://github.com/pptman/clarity_processor). Because they are not
> packaged it is difficult to use them. I see several options going
> forward:
> 
> 1) Create a PR to add setup.py and whatever else is necessary to make
> it a proper package.
> 2) Create a fork in MicronOxford and modify it however we want.
> 3) Add a new config option for the Clarity device that specifies a
> path to clarity_process.py, then load the module dynamically with
> importlib.util.
> 4) Don't do anything. Leave it to the the user to setup the
> clarity_process module.
> 
> Do you have any preferences? I like the first two options the best and
> I am happy to talk to Mark Neil about it.

Can we please solve this issue and get the working version of the clarity processing into the main branch.