python-microscope/microscope #122: undeclared dependency on PIL (Pillow)

carandraug, 2019-11-28:
The the `microscope.testsuite.devices` modules is dependent on PIL. This is used by the `ImageGenerator` class, which is used by the `TestCamera`, to add numbers to the camera.

We can add `PIL` to `setup.py` but I was thinking of maybe trying the import of PIL and simply not number the images if it's missing. Opinions?

mickp, 2019-11-28:
I guess image numbering isn't essential, but it can be really useful in diagnosing situations with multiple clients. It could be even more useful if we put it as a testing feature on the base class, so we could annotate an image from any camera - this could really help with diagnosing triggering and lost image problems. Either way, it could be an optional thing that is only available if you've installed PIL; I don't see that causing any problems, provided that we clearly document that PIL is required for the feature.

iandobbie, 2020-05-05:
PIL is also now used for the mosaic camera mode, I suggest that we just add it as a dependency.