python-microscope/microscope #128: ximea camera does not work

carandraug, 2019-12-12:
Support for the Ximea camera was added with PR #72. However, it seems that the code never worked. We had issue #107 where `get_sensor_shape` never worked, and now we have PR #126 where the whole trigger source never worked.

I have found that the version we have deployed on DeepSIM is completely different. It seems to have diverged on October 2018 so most of the implementation we have on master are not being used while there's a whole separate development branch on Nick's branch which is what we have been using.  We need to merge Nick changes back into master (and figure out how we ended up in this situation in the first place so it won't happen again).

juliomateoslangerak, 2019-12-18:
Indeed I just closed a related issue in BeamDelta that belongs here:
https://github.com/MicronOxford/BeamDelta/issues/15
I'm running the same camera model as yours and I probably have the latest ximea drivers as I just installed.
I basically found two issues.
- Sensor shape is retrieved through img.width and should be through the handle.get_width call. 
- Also there is a built in function to get an image as a numpy array. this one returns a uint8 array. This seems to solve my two problems with BeamDelta.

juliomateoslangerak, 2019-12-18:
I made a pull request #130 that fixes this issue and https://github.com/MicronOxford/BeamDelta/issues/15

carandraug, 2019-12-18:
@NickHallONBI can you help us on getting the diverged code that is running on the DeepSIM systems here? I'm starting to fear that this may not be specific to the ximea camera. Is the code that is actually ran somewhere on github? If so, which branch and which fork? For each of the DeepSIM systems.

NickHallONBI, 2019-12-18:
@carandraug @juliomateoslangerak The branch of microscope currently running on DeepSIM is the master branch of my fork (https://github.com/NickHallONBI/microscope.git). I have been busy this week and so have not got to comparing the differences between the two.

carandraug, 2019-12-19:
So you confirm that /all/ computers in the DeepSIM room are running the master branch of your fork? I will try to pick things from it.

juliomateoslangerak, 2019-12-19:
I looked into @NickHallONBI master branch and it indeed already integrates the changes I proposed in PR #130. So if you may ignore it. There are however some styling changes that are important. I'm a style fan I know.

NickHallONBI, 2019-12-20:
> So you confirm that /all/ computers in the DeepSIM room are running the master branch of your fork? I will try to pick things from it.

I checked the computers before I left. The DSP and SLM computers don't run microscope because neither of those are true microscope devices. The AOhost computer (the one actually connected to the Ximea camera) and the main computer both have my fork's master branch of microscope running. 

I dropped the ball and forgot to check the laser computer, but I suspect it is not running my master branch of microscope as I don't believe anyone has updated it any time in 2019. The last time I believe it had an update was to get the Sapphire laser to work on it.

carandraug, 2019-12-29:
I have picked the commits from the diverged developed in deepsim (NickHallONBI/master) and merged them back into master.

There's still some issues. At least the whole ROI stuff does not work. `_set_roi` stores the values in `self._roi` and `get_roi` returns them but nowhere else is `self._roi` is being used or set on the camera itself.

carandraug, 2020-01-17:
Having been working on the ximea camera now, the following features are missing:

* ROIs
* binning
* set image bitdepth
* readout mode
* sensor temperature

carandraug, 2020-01-20:
Trying to fix ROI support in the ximea camera, I have found a bug on `get_sensor_shape`. It uses xiapi's `Camera.get_width` and `Camera.get_height` methods.  However, those methods return the width and height of the image that will be returned which can be smaller than the sensor size in case we are doing ROIs. Looking through the docs, there does not seem to a way to get the real values so I've asked ximea about it.

carandraug, 2020-01-21:
> Looking through the docs, there does not seem to a way to get the real values so I've asked ximea about it.

ximea bug tracker is private, but the answer is that we can get sensor width via `width_max+offset_x_max` using the `get_width_maximum` and `get_offsetX_maximum` methods.  There's something similar for height.  The thing to note is that maximum width is based on the current state which is why we need to add offsetX so for example, if offsetX is set to zero and width is set to the sensor width, then the maximum of offsetX is zero since it can't be increased without first reducing the width.

Also tried to add binning but found that our ximea cameras don't have any support for it. Implementing it seems straightforward but I'm worried about things such as changing binning when there's already a ROI in place so I prefer to not do it until we get a camera where we can test it.

carandraug, 2020-01-21:
Pushed to master. Closing as fixed.

closed 2020-01-21 by carandraug