python-microscope/microscope #137: Issue with Aurox Clarity calibration image with Prime BSI

NickHallONBI, 2020-02-18:
We recently swapped  out the Andor Zyla we previously had on the Clarity for a Photometrics Prime BSI. The machine is running this branch of microscope, [NickHallONBI/mirao_stuff](https://github.com/NickHallONBI/microscope/tree/mirao_stuff), which has the microscope stuff for the Mirao52e DM as well as the latest commits from MicronOxford/microscope/master.

When starting up the deviceserver with the Clarity with the BSI as the slaved camera there are some errors in the log:

> 2020-02-18 16:41:37,340:Clarity (microscope.cameras.pvcam):WARNING:PID 12380: Skipping unsupported parameter PARAM_SMART_STREAM_EXP_PARAMS.
2020-02-18 16:41:37,341:Clarity (microscope.cameras.pvcam):WARNING:PID 12380: Skipping unsupported parameter PARAM_SMART_STREAM_DLY_PARAMS.
2020-02-18 16:41:37,346:Clarity (microscope.cameras.pvcam):WARNING:PID 12380: Skipping parameter PARAM_FRAME_BUFFER_SIZE: not supported in python.

On starting Cockpit, if you take a calibration image with no light source defined, the BSI behaves normally, the data is fetched and the calibration routine is run. However, if you define a light source (in our case, "ambient" with 100ms exposure time) you get this error:

> 2020-02-18 16:42:03,534:Clarity (microscope.cameras.pvcam):DEBUG:PID 12380: Fetched frame from circular buffer.
2020-02-18 16:42:04,264:Clarity (microscope.devices):ERROR:PID 12380: in _dispatch_loop:
Traceback (most recent call last):
  File "c:\microscope\microscope\devices.py", line 540, in _dispatch_loop
    self._send_data(client, self._process_data(data), timestamp)
  File "c:\microscope\microscope\devices.py", line 706, in _process_data
    return super()._process_data(data)
  File "c:\microscope\microscope\devices.py", line 507, in _process_data
    return functools.reduce(lambda x, f: f(x), self.pipeline, data)
  File "c:\microscope\microscope\devices.py", line 507, in <lambda>
    return functools.reduce(lambda x, f: f(x), self.pipeline, data)
  File "c:\microscope\microscope\filterwheels\aurox.py", line 164, in _c_process_data
    self._processor = ClarityProcessor(data)
  File "C:\clarity_processor\clarity_process.py", line 122, in __init__
    (dmidr, pos6r, diff6r) = self._find_penrose(posr)
  File "C:\clarity_processor\clarity_process.py", line 89, in _find_penrose
    dmid = np.sort(sep.ravel())[int(3.5 * n)]
IndexError: index 0 is out of bounds for axis 0 with size 0

It's also worth noting that no new data is fetched and displayed in the cockpit camera viewer either. If you then turn off the light source (and now once again have no light source defined) and try and take a calibration image, you get the same error as above except the last line is changed to:

> IndexError: index 3 is out of bounds for axis 0 with size 1

The "raw" and "difference" modes work fine with and without a light source being defined. 

It would appear that either the Clarity and/or the BSI are not being triggered in "calibration" mode when an light source is (or has been previously) defined.

mickp, 2020-02-18:
Cockpit should trigger whatever cameras are active even if a light source
is not active, but the exposure time will not be defined. For calibration,
the Clarity uses its own internal light source. You should activate the
"ambient" light source in cockpit to control exposure time.

Those messages from the BSI on startup are normal - they're warning you
that certain parameters are unsupported, either by the device or by our
python wrapper.

iandobbie, 2020-02-19:
Seems like with the ambient light source on there is an error, maybe that the camera doesn't actually return an image as it is swallowed by the calibration routine. Can we have fake image returned to prevent the error? Maybe a blank image, or even something saying calibration but that might be more effort than it is worth.

mickp, 2020-02-19:
An image is returned when calibration is complete. I think I saw opencv
errors in the original log, so perhaps the calibration routine is failing.

mickp, 2020-02-19:
Calibration is failing in ```_find_penrose``` because it is passed an argument, ```posr```, which is equal to zero. It looks like no spots are being found in the right-side image. I suspect that selecting a light source sets an exposure time which produces an image in which the spots are not clear - either they are buried in noise, or the image is saturated. Try selecting 'ambient' and changing its exposure time.

NickHallONBI, 2020-02-21:
Can confirm that is was a saturation issue. Tried the calibration image with 10ms exposure and that worked. A less esoteric error might be helpful, but this isn't a microscope issue. Closing this.

closed 2020-02-21 by NickHallONBI