python-microscope/microscope #139: Aurox clarity code and primeBSI camera.

iandobbie, 2020-02-24:
The prime BSI is supposed to return 16 bit data. The clarity process code appears to truncate this to 12 bit which then very easily saturates in the raw data.

carandraug, 2020-02-26:
I don't see any code on our side doing any truncation, so I'm guessing this is an issue on [clarity_processor](https://github.com/pptman/clarity_processor) itself.

Does this happens even on raw mode? Or does it only happen on difference mode?

mickp, 2020-02-26:
[clarity_processor](https://github.com/pptman/clarity_processor) converts incoming data to ```cv2.UMat```s which should preserve the full depth. For raw data, there is no conversion. Has the camera been set up to return 16-bit data in its settings? I think the power-on default may not be the full bit depth, as the maximum acquisition rate decreases with bit depth.

iandobbie, 2020-11-06:
I guess we need to test this further. I may have been running the camera in a mad mode, my recent experience with the Andor Zyla is it is easy to missconfigure this without realising it as there are so many settings.