python-microscope/microscope #140: Wrong images from Andor Zyla 5

VroniPfann, 2020-02-24:
@carandraug We swapped the shape of the numpy array, width by height. That way we can now see rectangular images, before only square images by default were possible.

carandraug, 2020-02-25:
The Zyla 5.5 has a rectangular sensor (2560x2160) and the acquired images were wrong. Changing the ROI for a square fixed the issue and then swapping the dimensions when creating the image buffer on SDK3 fixed it for good.

I have checked now that this was written for a Zyla 4.2 which has a square sensor (2048x2048) so I'm guessing this has just been always wrong and not something that the Andor SDK is doing different now.

carandraug, 2020-02-25:
Closing as fixed.

closed 2020-02-25 by carandraug