python-microscope/microscope #141: Add support for Thorlabs Stage NanoMax 300

VroniPfann, 2020-02-24:
It would be nice if the Thorlabs could be supported. We have a Thorlabs stage Nanomax 300 (https://www.thorlabs.com/thorproduct.cfm?partnumber=BSC203). This stage is motorised in x,y,z and has two rotational manual screws for Pitch and Yaw. In addition, it has analog signals for x,y,z.

iandobbie, 2020-02-26:
Do you have any documentation about the controllers command set? I had a look but all I could find was info about using it through thorlabs apt control software.

carandraug, 2020-02-26:
The controller is a BSC201 and seems like it can talk serial. The command set pdf is on this page https://www.thorlabs.com/software_pages/ViewSoftwarePage.cfm?Code=Motion_Control&viewtab=2 or [direct link to pdf](https://www.thorlabs.com/Software/Motion%20Control/APT_Communications_Protocol.pdf)

aurelien-barbotin, 2020-11-16:
It looks like the Kinesis software that controls most Thorlabs stage can be controlled via C#, which can itself be called via python. There is an example here which we might want to try: https://github.com/trautsned/thorlabs_kenesis_python

carandraug, 2020-11-16:
From Thorlabs website suggests that there's also C libraries (the low level DLL libraries):

> The Kinesis Software features new .NET controls which can be used by 3rd party developers working in the latest C#, Visual Basic, LabVIEW™ or any .NET compatible languages to create custom applications. Included are low level DLL libraries available for applications not expected to use the .NET framework.

Is using those low level DLL libraries much harder than the .NET library? This adds yet another dependency (and a dependency that's OS specific which makes testing with mocks more dfificult). It also adds another layer to debug when there's issues.

An example on how to use the C library, take a look at [microscope._wrappers.BMC](https://github.com/MicronOxford/microscope/blob/master/microscope/_wrappers/BMC.py) which is then used by the [microscope.mirror.bmc](https://github.com/MicronOxford/microscope/blob/master/microscope/mirror/bmc.py)

aurelien-barbotin, 2020-11-16:
Indeed, I missed that! I have actually never used DLL libraries but I agree that it is preferable to use less dependencies, so I will look into this

aurelien-barbotin, 2021-01-15:
Following your suggestions, I wrote some code to control the stage using ctypes. Shall I push it to the repository? I have written and tested code for a few basic methods (homing, moving to a given position etc.)

carandraug, 2021-01-15:
> Shall I push it to the repository? I have written and tested code for a few basic methods (homing, moving to a given position etc.)

Yes. Just push it to a branch on your fork and tell us the name of the branch (no need to open a pull request).

> The code needs to know the path to the DLL

Is the path to DLL really required in the code? Or is it enough for it to the on the path and let Python find it?

> and the serial number of the stage. How is that handled in microscope? Does it have to be hard coded?

The serial number is likely to be an argument to construct the class. It's the same as the `microscope.mirror.bmc` example I mentioned previously.

carandraug, 2021-02-01:
Hi @aurelien-barbotin I have started to review your implementation of the stage. The wrapper to the library seems fine. Do you also have the implementation of the actual stage interface and just forgot to add it to the commit or are you still working on that?

aurelien-barbotin, 2021-02-02:
Hi @carandraug , still working on it unfortunately. I hope I can finish it this week.

aurelien-barbotin, 2021-02-05:
Ok, I have a completed version of the stage interface. I could further test my code using a simulator provided by Thorlabs. All works fine except for the method to get the axis limits, which always returns 0. I hope it is only a bug in the simulator, I will try to test it on a real system but not sure I will manage to get that done.

The other thing I did not get to work is the conversion from device units to real units. There, the methods even returns an error when I try to use it. 

Last thing, there is a method to check that a given motion is finished, but they do not seem extremely reliable to me. I tried on the simulator and it sometimes freezes without me understanding why. I am anyway not sure how I would implement such methods in the interface class?