python-microscope/microscope #142: Issue while trying the Python Microscope on a xiB-64 camera

john-wigg, 2020-03-16:
Hello,
I tried to quickly run python Microscope on a ximea Camera using the following code

```
import microscope
from microscope.cameras.ximea import XimeaCamera
cam = XimeaCamera()

class DataClient(queue.Queue):
    def receiveData(self, *args):
        self.put(args)

buffer = DataClient()
cam = TestCamera()
cam.initialize()
cam.enable()
cam.set_client(buffer)

cam.soft_trigger()
img = buffer.get()
```

and I get the following error message in the terminal:

```
Traceback (most recent call last):
  File "testmicroscope.py", line 2, in <module>
    from microscope.cameras.ximea import XimeaCamera
ImportError: No module named ximea
```
my camera is a CB013MG-LX-X8G3 (xiB-64), I have the 4.19.21 version of the API package.

Thanks for the help

NickHallONBI, 2020-03-16:
Hi!

That looks like the API is not in the Python path. To confirm, have you moved the Python3 ximea package (found in the XIMEA\API\Python\v3 of the XIMEA folder the installer creates) into the Python path? If not, do so and check if the error persists

iandobbie, 2020-03-16:
The error seems more like it is an older version of microscope from before the ximea module was in the master branch.

Do you even have a file microscope/cameras/ximea.py from you microscope install directory?

carandraug, 2020-03-16:
> The error seems more like it is an older version of microscope from before the ximea module was in the master branch.

@john-wigg  are by any chance using Python 2? Python 2 is not supported since version 0.3.0 and there was no support for ximea cameras back then.

iandobbie, 2020-10-15:
@john-wigg Did you manage to solve this issue?