python-microscope/microscope #146: Add a test controller combining teststage and mosaic testcamera

iandobbie, 2020-05-05:
A test controller which combines the test stage to simulate stage movement and the testcamera mosaic mode to return images from the test mosaic image based on the stage coordinates. This allows simulation of the mosaic to work with no real hardware.

iandobbie, 2020-05-08:
I have started to implement this in branch https://github.com/iandobbie/microscope/tree/mosaicTestCamera
However I have a problem. The controller functionality works fine and will talk to cockpit by I now want to have the stage component have a link to the camera and be able to set the mosaic X pos and mosaic Y pos settings if the stage moves. The simulated camera then returns different segments of the mosaic image when it is triggered. 

The device server is doing something I don't understand and not allowing me to setup        
stage = StageIndirection(camera=camera)
which is a stage class which also accepts the camera object.   It returns:

2020-05-08 15:10:07,110:device-server (microscope.deviceserver):CRITICAL:PID 66082: name 'camera' is not defined

How do I do this? or the equivalent in another way.

iandobbie, 2020-05-08:
Now I understand thaT I need it in the __init__ function but I now have another problem that it cant get the device.axes.values() in the device server. Still struggling  through it.

carandraug, 2020-05-08:
```
2020-05-08 15:10:07,110:device-server (microscope.deviceserver):CRITICAL:PID 66082: name 'camera' is not defined
```

You're trying to subclass a `camera` class but that does not exist. The line that is causing the issue is the following: 

```
class StageIndirection(devices.StageDevice,camera):
```

I would recommend against using the device server until you have finished implementing this device. The device server will only add one other level of complexity and make debugging more difficult.  If you just try to run that module, the error would be obvious:

```
$ python3 microscope/controllers/testMosaicStageCamera.py
Traceback (most recent call last):
  File "microscope/controllers/testMosaicStageCamera.py", line 64, in <module>
    class StageIndirection(devices.StageDevice,camera):
NameError: name 'camera' is not defined
```

> The controller functionality works fine and will talk to cockpit by I now want to have the stage component have a link to the camera and be able to set the mosaic X pos and mosaic Y pos settings if the stage moves. 

I think this should be the one other way around. The x and y position are proprieties of the stage and the stage should not be setting it on the camera. Instead, the camera should be constructed with a stage as one of its argument. Then, when the camera wants to acquire an image it queries the stage for its position and returns the corresponding image subpanel.  I would image something like this:

```
class StageAwareCamera(CameraDevice):
    def __init__(self, stage: StageDevice, image: numpy.ndarray, **kwargs) -> None:
        ...
    def _fetch_data(self) -> return Optional[numpy.ndarray]:
        current_position = self._stage.position
        # Compute the indices for the image ndarray from the position
        return self.image[indices_from_stage_position]

# The controller is not necessary at all, a user could perfectly
# create its own camera and stage, it's only to make device server
# easier to use.
class CameraStageController(ControllerDevice):
    def __init__(self, image):
        stage = TestStage(....)
        camera = StageAwareCamera(self._stage, image)
        self._devices = {'stage' : stage, 'camera' : camera}

    @property
    def devices(self):
        return self._devices
```

iandobbie, 2020-05-08:
Ok, I will try that. There are of course other wrinkles but I want the basic functionality working first.

iandobbie, 2020-05-08:
On a side note I have now gone back to master rather than my testStage device branch and creating a microscope stage with 
    device(testdevices.TestStage, '127.0.0.1',8008)
but this generates an error about no limits, but I cant work out how to passing the limits in the stage setup.

iandobbie, 2020-05-08:
Oh the reason I have a deviceserver instance is I can then use cockpit and get the image data out to see if it is changing.

carandraug, 2020-05-08:
> but I cant work out how to passing the limits in the stage setup.

```
from microscope.devices import device, AxisLimits
from microscope.testsuite.devices import TestStage

DEVICES = [
    device(TestStage, 'localhost', 8009,
           conf={'limits' : {'X' : AxisLimits(0, 5000),
                             'Y' : AxisLimits(0, 5000)}}),
]
```

iandobbie, 2020-05-08:
Ok, I have started to try your approach but run into a different issue. Implementing a whole simulated camera from scratch is actually quite a lot of work, it would be much better to be able to utilise the existing testcamera, which I have already added a mosaic mode to. 

Is it possible to create a test camera, object and just intercept its ._fetch_data call to fist set the xy coords from the stage?

the reason I was trying to use the exisitng devices with just a tiny biut of added

carandraug, 2020-05-08:
> Is it possible to create a test camera, object and just intercept its ._fetch_data call to fist set the xy coords from the stage?

Sure.  You can even subclass the TestCamera and only re-implement the `_fetch_data` method, I think that should work.

iandobbie, 2020-06-16:
I finally made some more progress on this but I need help. 

My current state is at https://github.com/iandobbie/microscope/tree/stageAwareCam
I have set this up in combination with davids cockpit branch wip-572-microscope-stage. then the microscope config has 

import microscope.controllers.testStageAwareCamera as mosaicCam

DEVICES = [
 ....
    device(mosaicCam.CameraStageController, '127.0.0.1', 8008),
]

and cockpit depot file has
[testcontroller]
type: cockpit.devices.microscopeDevice.MicroscopeBase
uri: PYRO:CameraStageController@127.0.0.1:8008

[testStageCam]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
controller: testcontroller
controller.name: camera

[XY stage]
type: cockpit.devices.microscopeDevice.MicroscopeStage
controller: testcontroller
controller.name: stage
x-axis-name: x
y-axis-name: y
x-units-per-µm: 1
y-units-per-µm: 1

It starts up and almost works but crashes when I try and initialise the camera. 

The stage mostly works. Goto position works, double click on the mosaic to move the stage also works, but trying to run a mosaic doesn't. I think the stage doesn't return properly.

iandobbie, 2020-06-18:
I now have a working implementation in https://github.com/iandobbie/microscope/tree/stageAwareCam
It also needs a  very large tiff which is used to create the mosaic. I will upload this somewhere if anyone else wants to play. 

The relevant cockpit config is as above and requires my  branch https://github.com/iandobbie/cockpit/tree/pos-fix
as otherwise the microscope stage wont do mosaics. It also needs the camera to have a specific transform that is not yet set in config files.

iandobbie, 2020-06-18:
Issues with this branch:
mosaic image size is assumed, should be read
pixel size must be set for the relevant objective
Assumes mosaic mode is enum 6
no test for existence of image file, or if the load succeeds. 

Further extensions that would be good:
Method to select channel of input image so we can have multiple fluorescence channels
Use gaussian smoothing to simulate focus
Scale image so switching objectives can generate mosaic images with different sizes which work properly.

jstitlow, 2020-06-24:
Cockpit hangs during initialization. 

Environment (OSX Mojave 10.14.6):
1) created new Pandas environment (python v3.7.7 )
2) cloned cockpit repo
3) python setup.py install
4) Downloaded ftgl.  # but don't know what the hell to do with it
5) cloned microscope repo
6) pip install --no-index --editable microscope/
7) checked out ian/pos-fix and ian/stageAwareCam 
8) placed depot.py file in ~/Library/Application Support/cockpit
9) placed microscope-device.py in microscope/microscope # didn't see a src directory in there
10) placed mosaicimage.tif image in microscope/microscope/testsuite
11) run microscope    # seemed to work fine
12) run cockpit

[image]

Here is full error message:
Traceback (most recent call last):
  File "/Users/joshtitlow/src/cockpit/cockpit/__init__.py", line 149, in OnInit
    module = importlib.import_module(module_name)
  File "/usr/local/Cellar/python/3.7.7/Frameworks/Python.framework/Versions/3.7/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/Users/joshtitlow/src/cockpit/cockpit/gui/touchscreen.py", line 811, in <module>
    class LightToggleButton(SBitmapToggleButton):
  File "/Users/joshtitlow/src/cockpit/cockpit/gui/touchscreen.py", line 825, in LightToggleButton
    mask = wx.Mask(_bmp)
wx._core.wxAssertionError: C++ assertion "v == 0 || v == 3*0xFF" failed at /Users/robind/projects/bb2/dist-osx-py37/build/ext/wxWidgets/src/osx/core/bitmap.cpp(1563) in InitFromMonoBitmap(): Non-monochrome bitmap supplied

And here is console from running microscope:

(cockpit) Joshs-MacBook-Pro-6:microscope joshtitlow$ deviceserver microscope-devices.py
2020-06-23 16:39:21,025:TestLaser (root):DEBUG:PID 76301: Debugging messages on.
2020-06-23 16:39:21,029:TestCamera (root):DEBUG:PID 76299: Debugging messages on.
2020-06-23 16:39:21,035:DummyDSP (root):DEBUG:PID 76305: Debugging messages on.
2020-06-23 16:39:21,037:TestLaser (root):DEBUG:PID 76302: Debugging messages on.
2020-06-23 16:39:21,042:TestLaser (microscope.deviceserver):INFO:PID 76301: Device initialized; starting daemon.
2020-06-23 16:39:21,042:TestCamera (root):DEBUG:PID 76300: Debugging messages on.
2020-06-23 16:39:21,044:TestFilterWheel (root):DEBUG:PID 76303: Debugging messages on.
2020-06-23 16:39:21,044:TestLaser (microscope.deviceserver):INFO:PID 76301: Serving PYRO:TestLaser@127.0.0.1:8002
2020-06-23 16:39:21,044:TestCamera (PIL.PngImagePlugin):DEBUG:PID 76299: STREAM b'IHDR' 16 13
2020-06-23 16:39:21,045:TestCamera (PIL.PngImagePlugin):DEBUG:PID 76299: STREAM b'IDAT' 41 1216
2020-06-23 16:39:21,046:TestCamera (microscope.testsuite.devices):INFO:PID 76299: Initializing.
2020-06-23 16:39:21,046:TestFilterWheel (root):DEBUG:PID 76304: Debugging messages on.
2020-06-23 16:39:21,047:DummyDSP (microscope.deviceserver):INFO:PID 76305: Device initialized; starting daemon.
2020-06-23 16:39:21,048:DummyDSP (microscope.deviceserver):INFO:PID 76305: Serving PYRO:DummyDSP@127.0.0.1:8006
2020-06-23 16:39:21,050:TestLaser (microscope.deviceserver):INFO:PID 76302: Device initialized; starting daemon.
2020-06-23 16:39:21,050:TestLaser (microscope.deviceserver):INFO:PID 76302: Serving PYRO:TestLaser@127.0.0.1:8003
2020-06-23 16:39:21,052:TestCamera (PIL.PngImagePlugin):DEBUG:PID 76300: STREAM b'IHDR' 16 13
2020-06-23 16:39:21,052:TestCamera (PIL.PngImagePlugin):DEBUG:PID 76300: STREAM b'IDAT' 41 1216
2020-06-23 16:39:21,052:TestCamera (microscope.testsuite.devices):INFO:PID 76300: Initializing.
2020-06-23 16:39:21,053:CameraStageController (root):DEBUG:PID 76306: Debugging messages on.
2020-06-23 16:39:21,057:TestFilterWheel (microscope.deviceserver):INFO:PID 76304: Device initialized; starting daemon.
2020-06-23 16:39:21,057:TestFilterWheel (microscope.deviceserver):INFO:PID 76303: Device initialized; starting daemon.
2020-06-23 16:39:21,057:TestFilterWheel (microscope.deviceserver):INFO:PID 76304: Serving PYRO:TestFilterWheel@127.0.0.1:8005
2020-06-23 16:39:21,057:TestFilterWheel (microscope.deviceserver):INFO:PID 76303: Serving PYRO:TestFilterWheel@127.0.0.1:8004
2020-06-23 16:39:21,060:CameraStageController (PIL.PngImagePlugin):DEBUG:PID 76306: STREAM b'IHDR' 16 13
2020-06-23 16:39:21,060:CameraStageController (PIL.PngImagePlugin):DEBUG:PID 76306: STREAM b'IDAT' 41 1216
microscope/testsuite/mosaicimage.tif
2020-06-23 16:39:21,060:CameraStageController (microscope.testsuite.devices):INFO:PID 76306: Initializing.
2020-06-23 16:39:21,558:TestCamera (microscope.deviceserver):INFO:PID 76299: Device initialized; starting daemon.
2020-06-23 16:39:21,559:TestCamera (microscope.deviceserver):INFO:PID 76299: Serving PYRO:TestCamera@127.0.0.1:8000
2020-06-23 16:39:21,566:TestCamera (microscope.deviceserver):INFO:PID 76300: Device initialized; starting daemon.
2020-06-23 16:39:21,566:TestCamera (microscope.deviceserver):INFO:PID 76300: Serving PYRO:TestCamera@127.0.0.1:8001
2020-06-23 16:39:21,572:CameraStageController (microscope.deviceserver):INFO:PID 76306: Device initialized; starting daemon.
[... cut ...]
2020-06-23 16:55:31,502:TestLaser (microscope.testsuite.devices):INFO:PID 76301: Power set to 0.01.
2020-06-24 10:44:57,109:DummyDSP (microscope.testsuite.devices):INFO:PID 76305: Abort
2020-06-24 10:44:57,229:DummyDSP (microscope.devices):INFO:PID 76305: Current client is <Pyro4.core.Proxy at 0x1169a07d0; not connected; for PYRO:mui@127.0.0.1:7701>.
2020-06-24 10:44:57,231:TestLaser (microscope.testsuite.devices):INFO:PID 76301: Power set to 0.01.
2020-06-24 10:44:57,231:TestLaser (microscope.testsuite.devices):INFO:PID 76302: Power set to 0.01.

iandobbie, 2020-06-24:
Hi josh

This is an outstanding issue with wxpython 4.1.0 on Mac. We are trying to get to the bottom of it but if you do

Pip uninstall wxpython

Followed by

pip install wxpython==4.0.7

It should then work.

Ian

[image]<https://user-images.githubusercontent.com/5816582/85579414-da7f7080-b608-11ea-8262-3f300a5eabe5.png>

File "", line 1006, in _gcd_import
File "", line 983, in _find_and_load
File "", line 967, in _find_and_load_unlocked
File "", line 677, in _load_unlocked
File "", line 728, in exec_module
File "", line 219, in _call_with_frames_removed

jstitlow, 2020-06-24:
Cool, installing older version of wxpython worked. 

Now when I start cockpit, all of the windows are created but then quickly disappear. All that is left is blank screen:

[image]

Here is the readout from the microscope console in case that is helpful:

(cockpit) Joshs-MacBook-Pro-6:microscope joshtitlow$ deviceserver microscope-devices.py
2020-06-24 16:17:32,586:TestCamera (root):DEBUG:PID 86206: Debugging messages on.
2020-06-24 16:17:32,589:TestLaser (root):DEBUG:PID 86208: Debugging messages on.
2020-06-24 16:17:32,596:TestFilterWheel (root):DEBUG:PID 86209: Debugging messages on.
2020-06-24 16:17:32,596:TestLaser (root):DEBUG:PID 86207: Debugging messages on.
2020-06-24 16:17:32,597:TestCamera (root):DEBUG:PID 86205: Debugging messages on.
2020-06-24 16:17:32,597:TestCamera (PIL.PngImagePlugin):DEBUG:PID 86206: STREAM b'IHDR' 16 13
2020-06-24 16:17:32,598:TestCamera (PIL.PngImagePlugin):DEBUG:PID 86206: STREAM b'IDAT' 41 1216
2020-06-24 16:17:32,598:TestCamera (microscope.testsuite.devices):INFO:PID 86206: Initializing.
2020-06-24 16:17:32,600:TestFilterWheel (root):DEBUG:PID 86210: Debugging messages on.
2020-06-24 16:17:32,602:TestLaser (microscope.deviceserver):INFO:PID 86208: Device initialized; starting daemon.
2020-06-24 16:17:32,602:TestLaser (microscope.deviceserver):INFO:PID 86208: Serving PYRO:TestLaser@127.0.0.1:8003
2020-06-24 16:17:32,607:TestCamera (PIL.PngImagePlugin):DEBUG:PID 86205: STREAM b'IHDR' 16 13
2020-06-24 16:17:32,608:TestCamera (PIL.PngImagePlugin):DEBUG:PID 86205: STREAM b'IDAT' 41 1216
2020-06-24 16:17:32,608:TestCamera (microscope.testsuite.devices):INFO:PID 86205: Initializing.
2020-06-24 16:17:32,609:TestLaser (microscope.deviceserver):INFO:PID 86207: Device initialized; starting daemon.
2020-06-24 16:17:32,609:TestFilterWheel (microscope.deviceserver):INFO:PID 86209: Device initialized; starting daemon.
2020-06-24 16:17:32,609:TestLaser (microscope.deviceserver):INFO:PID 86207: Serving PYRO:TestLaser@127.0.0.1:8002
2020-06-24 16:17:32,609:TestFilterWheel (microscope.deviceserver):INFO:PID 86209: Serving PYRO:TestFilterWheel@127.0.0.1:8004
2020-06-24 16:17:32,611:TestFilterWheel (microscope.deviceserver):INFO:PID 86210: Device initialized; starting daemon.
2020-06-24 16:17:32,612:TestFilterWheel (microscope.deviceserver):INFO:PID 86210: Serving PYRO:TestFilterWheel@127.0.0.1:8005
2020-06-24 16:17:32,612:CameraStageController (root):DEBUG:PID 86212: Debugging messages on.
2020-06-24 16:17:32,614:DummyDSP (root):DEBUG:PID 86211: Debugging messages on.
2020-06-24 16:17:32,619:CameraStageController (PIL.PngImagePlugin):DEBUG:PID 86212: STREAM b'IHDR' 16 13
2020-06-24 16:17:32,619:CameraStageController (PIL.PngImagePlugin):DEBUG:PID 86212: STREAM b'IDAT' 41 1216
microscope/testsuite/mosaicimage.tif
2020-06-24 16:17:32,619:CameraStageController (microscope.testsuite.devices):INFO:PID 86212: Initializing.
2020-06-24 16:17:32,623:DummyDSP (microscope.deviceserver):INFO:PID 86211: Device initialized; starting daemon.
2020-06-24 16:17:32,623:DummyDSP (microscope.deviceserver):INFO:PID 86211: Serving PYRO:DummyDSP@127.0.0.1:8006
2020-06-24 16:17:33,115:TestCamera (microscope.deviceserver):INFO:PID 86206: Device initialized; starting daemon.
2020-06-24 16:17:33,116:TestCamera (microscope.deviceserver):INFO:PID 86206: Serving PYRO:TestCamera@127.0.0.1:8001
2020-06-24 16:17:33,119:TestCamera (microscope.deviceserver):INFO:PID 86205: Device initialized; starting daemon.
2020-06-24 16:17:33,119:TestCamera (microscope.deviceserver):INFO:PID 86205: Serving PYRO:TestCamera@127.0.0.1:8000
2020-06-24 16:17:33,128:CameraStageController (microscope.deviceserver):INFO:PID 86212: Device initialized; starting daemon.
2020-06-24 16:17:33,128:CameraStageController (microscope.deviceserver):INFO:PID 86212: marshal was removed from accepted serializers
2020-06-24 16:17:33,129:CameraStageController (microscope.deviceserver):INFO:PID 86212: Serving PYRO:CameraStageController@127.0.0.1:8008
2020-06-24 16:17:42,447:DummyDSP (microscope.testsuite.devices):INFO:PID 86211: Abort
2020-06-24 16:17:42,464:DummyDSP (microscope.devices):INFO:PID 86211: Current client is <Pyro4.core.Proxy at 0x10eda3210; not connected; for PYRO:mui@127.0.0.1:7701>.
2020-06-24 16:17:42,466:TestLaser (microscope.testsuite.devices):INFO:PID 86207: Power set to 0.01.
2020-06-24 16:17:42,466:TestLaser (microscope.testsuite.devices):INFO:PID 86208: Power set to 0.01.

iandobbie, 2020-06-26:
This is a new one on me. This is I think a cockpit issue, rather than microscope but I don't know what. I am on pythion 3.7.7 and macos 10.14.6 with the wxpython 4.0.7 as I suggested you install. I am not using panda's though. Is there anything in the cockpit console window? Looks to me like its lost some vital libraries and just not starting up enough to get anywhere.

carandraug, 2020-06-26:
> This is a new one on me. This is I think a cockpit issue, rather than microscope but I don't know what. [...]

I'm pretty sure this was fixed by MicronOxford/cockpit@d4e890bbdaa6aa1c979b61e36da3a7f7659e77c6 . @iandobbie maybe merge master into the pos-fix branch to get other fixes.

jstitlow, 2020-06-26:
thanks, @iandobbie  I'm happy to check it out if you don't mind merging those fixes.

iandobbie, 2020-06-26:
Sorry tried to rebase on master but things are a mess, my origin brach seems to be out of sync with the local branch. I will investigate and let you know where I have got to.

iandobbie, 2020-06-29:
I have rebased my cockpit branch on the current master and it seems to work fine for me.

iandobbie, 2020-06-29:
It is worth putting in the camera transform as without this the images dont line up as they are rotated. The current depot.conf entries are:

`[testcontroller]
type: cockpit.devices.microscopeDevice.MicroscopeBase
uri: PYRO:CameraStageController@127.0.0.1:8008

[testStageCam]
type: cockpit.devices.microscopeCamera.MicroscopeCamera
controller: testcontroller
controller.name: camera
transform: (1,1,0)

[XY stage]
type: cockpit.devices.microscopeDevice.MicroscopeStage
#uri: PYRO:TestStage@127.0.0.1:8008
controller: testcontroller
controller.name: stage
x-axis-name: x
y-axis-name: y
x-units-per-µm: 1
y-units-per-µm: 1
`

iandobbie, 2020-07-01:
Updated to solve 

> Issues with this branch:
> mosaic image size is assumed, should be read

This is now fixed

> pixel size must be set for the relevant objective

Not sure how to do this as cockpit does not pass the pixel size. 

> Assumes mosaic mode is enum 6

This is fixed as it now adds the new image generator function to the end of the list and then selects the last entry on the list. 

> no test for existence of image file, or if the load succeeds.

Now logs a message and generates a small 0 filled image

> Further extensions that would be good:
> Method to select channel of input image so we can have multiple fluorescence channels

It now adds a "mosaic channel" setting which selects from the image channels. Should implement a filter wheel, or maybe config parameter so this can be applied to multiple cameras a provide different images. 

> Use gaussian smoothing to simulate focus
> Scale image so switching objectives can generate mosaic images with different sizes which work properly.

Neither of these currently working. 

If people are testing this please pull the latest version of the https://github.com/iandobbie/microscope/tree/stageAwareCam

iandobbie, 2020-07-01:
Added Z stage and an increasing gaussian blur as the stage moves away from 0

jstitlow, 2020-07-01:
I've now got Cockpit up and running now on my Mac laptop, thanks!

Mosaic simulation seems ok to me but with some issues but if I run it twice in a session, the second time it runs along a wide perimeter:

[image]

iandobbie, 2020-07-01:
I think you just have a saturated image. What data range do you get in the camera view window? Can you see the image there? Also what happens if you click the rescale tiles in the mosaic window?

iandobbie, 2020-07-01:
See the black edge of the image here that happens once you reach the edge of the image position. 

[image]

iandobbie, 2020-07-02:
This branch works for me, but requires several other changes to be merged, can we please get them merged so the changes needed here can be sorted and a clean set of commits generated. I am very keen to include this in the master branch as it is a brilliant demonstartion tool.

jstitlow, 2020-07-02:
It is rather calming to watch it go round and round. Works for me now, I was giving devices.py bad directions to the mosaicimage.tif file.

iandobbie, 2020-07-02:
> It is rather calming to watch it go round and round. Works for me now, I was giving devices.py bad directions to the mosaicimage.tif file.

This should now generate a log message.

iandobbie, 2020-07-10:
I have merged the stageAwareCamera controller into the current master branch with a new branch https://github.com/iandobbie/microscope/tree/WIP_stagecam which is just the one additional file, btu it also needs a large mage to generate the mosaic from.

iandobbie, 2020-09-08:
I am very keen to merge this. Is there any reason not to?

I have tidied up the code a tiny bit to remove the option for generating random errors, which still exisits in the standard fake cameras so seems a bit redundant.

iandobbie, 2020-09-08:
The one outstanding issue is it would be nice to return a scaled image depending on the objective but this needs a mechanism to get the pixel size info to the camera.

carandraug, 2020-12-15:
I made some changes to this, rebased by squash, and pushed to master. This is commit f8782a2 . I have added the required cockpit and microscope configurations to the MicronOxford/configs repo.

closed 2020-12-15 by carandraug