python-microscope/microscope #149: Installation fails at hidapi

juliomateoslangerak, 2020-06-09:
Hi,

I'm trying to do a fresh installation on my mac of cockpit and i fails at the installation of hidapi.
So here is what I do:
- new directory
- create a venv based on python 3.6 (I believe it also fails under 3.7 but I will check that again)
- clone the git repo 
- run python setup.py install 

It starts to spin and install stuff until hidapi. Here is the section with the error:

`Installed /Users/julio/PycharmProjects/Cockpit_microscope/venv/lib/python3.6/site-packages/Pillow-7.1.2-py3.6-macosx-10.9-x86_64.egg
Searching for hidapi
Reading https://pypi.org/simple/hidapi/
Downloading https://files.pythonhosted.org/packages/7c/a0/d5ca6f191c8860a4769ba19448d2b2d6b3e2ca2c30aa61bb96a3f6bd25ba/hidapi-0.9.0.post2.tar.gz#sha256=a71dd3c153cb6bb2b73d2612b5ab262830d78c6428f33f0c06818749e64c9320
Best match: hidapi 0.9.0.post2
Processing hidapi-0.9.0.post2.tar.gz
Writing /var/folders/2q/q8rfyc1d7xjbt2jtmbtb93gc0000gn/T/easy_install-o5qy4ckq/hidapi-0.9.0.post2/setup.cfg
Running hidapi-0.9.0.post2/setup.py -q bdist_egg --dist-dir /var/folders/2q/q8rfyc1d7xjbt2jtmbtb93gc0000gn/T/easy_install-o5qy4ckq/hidapi-0.9.0.post2/egg-dist-tmp-tpas6wtv
WARNING: The wheel package is not available.
/private/var/folders/2q/q8rfyc1d7xjbt2jtmbtb93gc0000gn/T/easy_install-o5qy4ckq/hidapi-0.9.0.post2/.eggs/Cython-0.29.19-py3.6-macosx-10.9-x86_64.egg/Cython/Compiler/Main.py:369: FutureWarning: Cython directive 'language_level' not set, using 2 for now (Py2). This will change in a later release! File: /private/var/folders/2q/q8rfyc1d7xjbt2jtmbtb93gc0000gn/T/easy_install-o5qy4ckq/hidapi-0.9.0.post2/hid.pyx
  tree = Parsing.p_module(s, pxd, full_module_name)
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDLib.h:31:0,
                 from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDManager.h:29,
                 from hidapi/mac/hid.c:25:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDDevice.h:298:34: error: unknown type name 'dispatch_block_t'
                                  dispatch_block_t               handler)
                                  ^
In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDLib.h:36:0,
                 from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDManager.h:29,
                 from hidapi/mac/hid.c:25:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDQueue.h:297:33: error: unknown type name 'dispatch_block_t'
                                 dispatch_block_t                handler)
                                 ^
In file included from hidapi/mac/hid.c:25:0:
/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/IOKit.framework/Headers/hid/IOHIDManager.h:295:33: error: unknown type name 'dispatch_block_t'
                                 dispatch_block_t                handler)
                                 ^
hidapi/mac/hid.c: In function 'create_device_info':
hidapi/mac/hid.c:487:3: error: 'for' loop initial declarations are only allowed in C99 or C11 mode
   for (CFIndex i = 0; i < CFArrayGetCount(usage_pairs); i++) {
   ^
hidapi/mac/hid.c:487:3: note: use option -std=c99, -std=gnu99, -std=c11 or -std=gnu11 to compile your code
error: Setup script exited with error: command 'gcc' failed with exit status 1
`

carandraug, 2020-06-09:
hidapi is a microscope dependency so I'm transferring this issue to the microscope repo. This is probably a hidapi issue anyway, and there will be little we can do about it other than removing the dependency.

Can you report this to them? You should be able to reproduce this by simply `pip install hidapi`. I have already found a few issues on their tracker and they recommend against using anaconda and virtual envs. So maybe try without those before reporting it to them.

juliomateoslangerak, 2020-06-09:
Indeed I can reproduce with just `pip install hidapi`.

carandraug, 2020-07-12:
This is an issue either or Julio's setup or on hidapi. Closing as invalid.

closed 2020-07-12 by carandraug