python-microscope/microscope #151: device-server crashes on startup with "'NoneType' object has no attribute 'loader'"

microphysics, 2020-06-25:
I tried different operating systems, python versions and microscope-versions, all gave the same error:

python3 -m microscope.deviceserver microscopetest.cfg
2020-06-25 11:48:36,728:device-server (__main__):CRITICAL:PID 49124: 'NoneType' object has no attribute 'loader'

The content of ~/microscopetest.cfg is:
from microscope.devices import device

from microscope.testsuite.devices import TestCamera
from microscope.testsuite.devices import TestLaser

DEVICES = [
    device(TestCamera, '127.0.0.1', 8005),
    device(TestLaser, '127.0.0.1', 8006),
]

carandraug, 2020-06-25:
I think the issue is because the file needs the `.py` file extension (well, it needs `importlib.machinery.SOURCE_SUFFIXES`). I'm looking into fixing this now, thank you for the report.

carandraug, 2020-06-25:
This should be fixed (see 6043742952ae) which will be part of the next release (0.6.0). In the mean time, you can use the development version from git, or rename the file to have the `*.py` extension.

It is possible that there is other issue beyond the file extension but I can't replicate it. Can you let us know if that fixes it and close this issue?

microphysics, 2020-06-25:
Renaming the config file to xyz.py results in a different error:

mv microscopetest.cfg microscopetest.py
python -m microscope.deviceserver microscopetest.py
2020-06-25 12:44:47,574:device-server (__main__):CRITICAL:PID 91499: No module named 'PIL'

The current git version gives the same error:
~/git/microscope/microscope $ python deviceserver.py ~/microscopetest.py
2020-06-25 13:34:46,965:device-server (__main__):CRITICAL:PID 127285: No module named 'PIL'

carandraug, 2020-06-25:
That was because Pillow was not listed as a dependency. We recently added it (see issue #122). Glad it's all working now. Closing as fixed.

closed 2020-06-25 by carandraug

iandobbie, 2020-06-26:
As David's reply doesn't explain what you need to do, not sure if he communicated this to you separately. You need to explicity install pillow using

`pip install pillow`

microphysics, 2020-06-26:
Hey Ian, I am not sure how image and PIL/pillow are related in python, but I think I already had everything installed. Google/Stackoverflow told me to import image first*, which I did, and then I could start the deviceserver.

I just tried to start the deviceserver without importing image first and it works, even though I can't remember any changes to my python installation since yesterday.

Thanks.

https://stackoverflow.com/questions/8863917/importerror-no-module-named-pil

> On some installs of PIL, You must do
>import Image
>instead of import PIL (PIL is in fact not always imported this way). Since import Image works for you, this means that you have in fact installed PIL.

carandraug, 2020-06-26:
> [...] I am not sure how image and PIL/pillow are related in python [...]

"Pillow" is a Python package distribution (what you search for in PyPI and install via pip); `PIL` is a Python package in the "Pillow" package distribution (some package distributions have more than one Python package); `Image` is a class in the `PIL` package.

> I just tried to start the deviceserver without importing image first and it works, even though I can't recall any changes to my python installation since yesterday.

You probably installed "Pillow" yesterday. You shouldn't have need to import anything, just to install "Pillow".