python-microscope/microscope #156: XY stages for the prior controller

carandraug, 2020-07-13:
We need to add support for prior stages. There was PR #99 but that implemented them as individual stage device instead of devices in the prior controller. The prior controller has already been implemented but only has support for filterwheels.

carandraug, 2020-07-13:
I have a `156-prior-controller-stage` branch that is supposed to implement this from back in February.  There were some minor things missing which I now filled in but this was all from 5 months ago so I may be misremembering some bits.  With the current situation I won't be able to go and test it with hardware.  I'm leaning towards merging this anyway.  Opinions?

carandraug, 2020-07-28:
@dstoychev has tested this and seems to be working. It timeouts in some cases but turns out that the stage has some encoder issues. We will wait until that is fixed to check if there is an issue with the code.

Also, as discussed on Monday meeting, this device disables the servo. This is because we don't want to risk having the servo adjust position in the middle of one exposure, it's better to have a slow drift over the time course of an experiment. But maybe we should add it via the setting scheme.

Also discussed on Monday, we don't home the stage with the home command. Instead, when the stage is enebled we manually move to the corners to get the stage limits. Ian says that we need to call home (`SIS` command) to get absolute positions instead of relative since the last reboot. The documentation seems to suggest otherwise but it's confusing and often incorrect anyway. We need testing to confirm this.

iandobbie, 2020-08-10:
I should just say that my experience of having to home the device to get absolute positioning may be due to using an older proScan II controller whereas Danny is using the proScan III. We should check exactly what is needed with the current controller.

iandobbie, 2020-10-14:
@dstoychev is the prior controller code working with the repaired stage? If so we should pull it and close the issue.

iandobbie, 2022-06-08:
I think this is working so should be closed. If we need to add a home on enable command then it should follow the convention that I just started with the ludl stage. 

1) Implement need_homed to be true if homing is required

2) implement a home function that homes the stage, sets limits if needed etc.

closed 2022-06-08 by iandobbie