python-microscope/microscope #158: Issue finding uid for linkamCMS stage.

iandobbie, 2020-08-17:
The FloatingDeviceMixin now requires a uid to register even a single device. This presents a problem about how do you find the magic serial number of the physical hardware.

Updating the Oxford CryoSIM software I had two such devices, the andor ixon which at least had the serial number in the previous logs. The linkam stage though, I have no idea as to its serial number or how to find it. I thinkt he code is broken as it was previously returned as "uid of device is   ". I have managed to get it to work by using uid='' which seems like a terrible hack. I will try to find out what the uid actually is.

carandraug, 2020-08-17:
> The FloatingDeviceMixin now requires a uid to register even a single device. This presents a problem about how do you find the magic serial number of the physical hardware.

Is this not a label on the physicial device? All the ones I have handled have it printed somewhere.

> [...] I thinkt he code is broken as it was previously returned as "uid of device is " [...]

Reading the docuentation in the linkam module, I found:

```
NOTE: this module does not run correctly with python optimisations in
use. When invoked with python -O, there seem to be issues with accessing
ctypes objects.
[...]
  * get_id returns an empty string, not the device serial number.
```

so this seems like a known issue.

> [...] I have managed to get it to work by using uid='' which seems like a terrible hack. I will try to find out what the uid actually is.

I found this on the code:

```
        # The uid is used to set serialNumber on the info object. The docs
        # suggest that an OpenComms message should open a connection to the
        # device with that serial number; with only one stage attached, it
        # appears that OpenComms ignores the value of serialNumber.
```

which suggests that the linkam stage can connect to a specific device given the serial number which means that it shouldn't need to subclass from `FloatingDeviceMixIn`.

iandobbie, 2023-09-22:
This works with a single stage and an empty serial number. I suggest that we document this and close the issue as there doesn't seem much milage in keeping open a 3 year old issue with a work around and no other proposed solution.