python-microscope/microscope #162: Need to implement a digital io device and logger.

iandobbie, 2020-09-03:
Asynchronous digital IO for flip mirrors and similar. Currently there are two cockpit modules which do this the piDIO module and the NI6036. Both of these modules control digital devices but with no high speed or defined timing. They also provide temperature (and possibly other analouge signal) logging. 

A digital device needs:

read state
set state

Possibly also requires reading a setting multiple lines at once.

Signal logging is a bit more complicated as we need to decide if the values are pushed or pulled.

carandraug, 2020-09-03:
The NI6036 class in cockpit was removed in MicronOxford/cockpit@e2264fe2b045a361c0e0a2b930e5fc8a5808b825

There is also the Redpitaya Server class on https://github.com/MicronOxford/RedPitaya-DSP/ (but note its different implementations on the different branches of that project as mentioned on MicronOxford/RedPitaya-DSP#10)

iandobbie, 2022-05-17:
I have pointed the cockpit issue on PIDIO startup issues here as pidio should be totally moved out of cockpit and into microscope with a new digital IO api and also I think temperature and other scalars leading to the value logger. I will investigate should be quite easy to be honest.

iandobbie, 2023-02-22:
This issue is completely covered by #270 and early related code. We now have an abstract class, a simulation and a concrete implementation on the Raspberry Pi. It does debug, land logvalueviewer as well.

closed 2023-02-22 by iandobbie