python-microscope/microscope #167: support PiCam (Raspberry Pi Camera)

carandraug, 2020-09-17:
Support the Raspberry Pi Camera (PiCam). Requested by @iandobbie . Initial work on iandobbie's "picam" and "picam-devel" branches.

carandraug, 2020-12-11:
@beniroquai : adding you to this issue, since you're planning on implementing support for the picamera first. As we spoke, Ian's most advanced branch to support the picamera was the `picam-devel`. I just merged master into it and made some adjustments to fit with the current microscope. I suggest you start with the `picam-devel` branch that it's currently on my fork.

iandobbie, 2020-12-11:
Hi everyone.

I am away with no computer this weekend. My memory is the software triggering worked fine but I planned to implement a hardware trigger via an interrupt on a gpio pin. In Theory this is  easy but I got distracted by other stuff before I actually implemented it.

Also this is all likely 2-3 years ago so the OS may have moved a little underneath it. Also might be wor to thinking about the new hq camera but we don’t yet have one of these.

Ian

carandraug, 2021-02-03:
@beniroquai : just checking, are you still planning on implementing the PiCAM in Microscope?

iandobbie, 2021-03-11:
I have done some work on this and got the basic framework in place but I now need some help. The current code is in my repository in wip-picam

I have a trigger which is setup to generate an interrupt from one of the GPIO pins. This currently calls:

    def HW_trigger(self,channel):
        '''Function called by GPIO interupt, needs to trigger image capture'''
        print ('PiCam HW trigger')

I want this function to capture an image, with some thing like  the software trigger code

            with picamera.array.PiYUVArray(self.camera) as output:
                self.camera.capture(output, format='yuv', use_video_port = False)
                #just return intensity values                                   
                return(output.array[:,:,0])

But not return the array but I guess put it on a queue which _fetch_data then empties, but I know nothing about how best to do queues in python.

beniroquai, 2021-03-12:
Hey, sorry for late reply. I was trying to play with `microscope`, but had troubles to get the GUI up and running on my machine. I'll definitely come back to it later again. For now I was able to hack a GBRL stage (CNC machine/3D printer), Lasers and Lenses as well as the `picamerax` from Bowman et al. into the openflexure-server system. 
The device adapters should be fairly simple to port again. The GBRL protocol caused me some problems. I'll try to merge the branch and see what I can do on my end here. Compared to the ordinary `picamera` package, the `picamerax` has more degrees of freedoms in terms of RAW acquisitions, selecting the gain and flat fielding.

iandobbie, 2021-08-19:
I have now implemented a bare bones driver that does software and hardware triggering. With a test setup I can do software triggering and hardware trigger from a red pitaya. Deals with exposure times, and all standard camera settings (rotation, roi etc...). I will add it to the main branch but it still needs support for 

a) the various camera settings
b) different camera modes, image sizes etc....
c) generally tidied up to read GPIO pin from config, etc....

carandraug, 2021-08-19:
Are you sure the code works? I see it uses `self._logger` which has been removed from the base class almost two years ago (with 8f2d2fdb1ac1bc6cae2dd). It is also missing a series a few of abstract methods and implements some that are no longer used.

iandobbie, 2022-02-21:
Hi am trying to get back to this again. I have merged my latest version with the current microscope main branch and have the software triggering working. I need to test the hardware trigger. As to the previous comment it now uses _logger which the other camera drivers also seem to use so I assume this is ok, also runs fine with main branch, just replacing the picam.py file.

iandobbie, 2022-02-22:
I have now tested with HW trigger and it still works with that. My branch is at https://github.com/iandobbie/microscope/tree/picam-update

I will keep working on this to add some of the picam config options.

carandraug, 2023-01-06:
@iandobbie already pushed this a few months ago. Closing as fixed.

closed 2023-01-06 by carandraug