python-microscope/microscope #168: Deepstar laser module spams the console.

iandobbie, 2020-10-06:
The deepstar laser code produces a console message every few seconds which is stupid. Needs to be suppressed in general usage.

carandraug, 2020-10-06:
Most of the devices, the deepstar laser included, use the info level for many of its messages. I guess we could change them to debug.

However, I think the root cause is that the device server changes the default logging level at start to debug (well, if `__debug__` is true which typically is). We should probably have a command line option for this.

carandraug, 2020-11-20:
I just set up a test deformable mirror to try Nick's aotools. This issue makes it impossible to get anything out of the logs, there's just too much stuff happening there. We can't default to debug log level.

carandraug, 2021-01-10:
I have added an `--logging-level` command line option to `device-server` and made it default to info. I'm not sure about it, I can also think of many reasons to have it default to warning. The thing that made me chose info in the end, was that printing the device URI is obviously info, and that's something that seems sensible to print by default. Also, if default is set to warning, nothing is printed and the feeling is that nothing is happening. The old unix silence is golden rule makes more sense to command line tools to be used interactively which is not the case of the device-server (which is more akin to a daemon).

Closing as fixed but I think its worth having a look at messages being logged and thinking twice about their level. There's some stuff which I think would fit debug better than info.

closed 2021-01-10 by carandraug

carandraug, 2021-04-21:
> [...] Closing as fixed but I think its worth having a look at messages being logged and thinking twice about their level. There's some stuff which I think would fit debug better than info.

I just did it for the `DeepstarLaser` which is what triggered this issue. The message that kept spamming the console was "Are we on? [S1]" which was info level. This was emitted every time `get_is_on` is called which Cockpit does every second or so.