python-microscope/microscope #169: Ximea temperature sensors.

iandobbie, 2020-10-15:
The xiC camera we have doesn't inilialize in the current code as the 
                self._handle.set_temp_selector(temperature_selector)
with temperature_selector = "XI_TEMP_IMAGE_SENSOR_DIE" 
returns 
ximea.xiapi.Xi_error: ERROR 100: Unknown parameter

but the except clause is 
     except:
                if getattr(err, "status", None) == 12:  # Not supported
I cant find a description of the error codes in the ximea API manual so I have no idea where the 12 came from.

Should we have two tests ==12 || ==100, or replace the 12 with 100?

carandraug, 2020-10-15:
The errors are described in Python at `ximea.xidefs.ERROR_CODES` and in C on the `XI_RET` enum on the `xiApi.h` header file:

```
ERROR_CODES = {    
...
    12: "Not supported",
...
    100: "Unknown parameter",
```

It is odd you're getting "Unknown parameter". I guess that unknown parameter means the library does not know what to do with "XI_TEMP_IMAGE_SENSOR_DIE" while "Not supported" means that the physical devices does not have that sensor. What is the value of `ximea.xidefs.XI_TEMP_SELECTOR` on your installation?

iandobbie, 2020-10-15:
{'XI_TEMP_IMAGE_SENSOR_DIE_RAW': c_uint(0), 'XI_TEMP_IMAGE_SENSOR_DIE': c_uint(1), 'XI_TEMP_SENSOR_BOARD': c_uint(2), 'XI_TEMP_INTERFACE_BOARD': c_uint(3), 'XI_TEMP_FRONT_HOUSING': c_uint(4), 'XI_TEMP_REAR_HOUSING': c_uint(5), 'XI_TEMP_TEC1_COLD': c_uint(6), 'XI_TEMP_TEC1_HOT': c_uint(7)}

>>> ximea.__version__
'4.21.18'

Tried explicitly opening the camera, opening the device and then calling _handle.set_temp_selector with ..._DIE or ..._DIE_RAW and found the same  result. 

Further with the DIE_RAW sensor I get 0.0 return to a get_sensor_temperature() call whereas if I set the sensor to XI_TEMP_SENSOR_BOARD I get a much more reasonable 56.0

iandobbie, 2020-10-16:
I should clarify the comment above, from a python prompt I also get error 100 when I try to set the sensor to ..._DIE but it works with ..._DIE_RAW. However the DIE_RAW seems to return rubbish data. It seems like the SENSOR_BOARD is the only sensor which is present and returns real data on the xiC MC023MG-SY-UB camera I have here.

carandraug, 2020-10-21:
I tried this on Linux with the same camera and I can replicate the issue. Selecting the temperature sensor `"XI_TEMP_IMAGE_SENSOR_DIE"` raises error code 100 ("Unknown parameter"). But not all unsupported temperature sensors raise that, `"XI_TEMP_INTERFACE_BOARD"` raises error code 12 ("Not supported").  Maybe there's an issue on xiapi, or maybe the two errors have different meaning. I have reported this upstream. Their issue tracker is not public, but my issue number is [56130](https://desk.ximea.com/tickets/56130)

Anyway, whatever the difference is, we need to handle both error code 12 and 100 as you suggested so I pushed that fix in 8d33a4c77171155027ff1e72189f91c041ceb94f.

The other issue is that `"XI_TEMP_IMAGE_SENSOR_DIE_RAW"` seems to not return anything useful, only `0.0`. I'm not sure how to address that. The order of temperature sensors to pick was chosen with Mick, the plan being that we try whatever is closest to the sensor. Changing that order because in this specific camera it doesn't work would prevent it from being chosen in cameras that it does work.

carandraug, 2020-10-26:
Ximea confirmed the issue that they use different error codes for temperature sensors that don't work, and that they plan on fixing it on a later version.

> thank you for notifying us about this ambiguity. Each enumerator item which returns error is not supported by this camera model, but we will unify the error prints.

iandobbie, 2020-10-26:
Once it’s documented it’s not a bug it’s a feature!

carandraug, 2020-10-27:
Upstream has also now confirmed that with the XiC camera, selecting `"XI_TEMP_IMAGE_SENSOR_DIE_RAW"` should be raising an error as this sensor is not available. They'll investigate where the issue is and why we get temperature measurements of zero. In the mean time, they recommend using `"XI_TEMP_SENSOR_BOARD"` as that "temperature sensor is located right next to the image sensor and the value should be very close to the die temperature value".

carandraug, 2020-11-03:
@carandraug wrote:
> Ximea confirmed the issue that they use different error codes for temperature sensors that don't work, and that they plan on fixing it on a later version.

They have fixed this on version 4.21.19. Instead of raising "Not supported" in some cases and "Unknown parameter" in others, it should now return "Unknown parameter" for unsupported sensors. Their reply:

> it has been fixed, now all unsupported temp. selector items return Error 100 - unknown parameter.

Why they didn't choose the "Not supported" error for the case of "unsupported" is a mistery. Their reply when asked about it:

> I admit that this particular error description is not very intuitive and we are planning to review all error messages in the future.

Hopefully this won't break much if they ever do such a review (I guess they just did it to shut me off).

We are still waiting on a update that fixes our xiC not raising an issue when selecting `"XI_TEMP_IMAGE_SENSOR_DIE_RAW"`.

closed 2023-03-09 by iandobbie

carandraug, 2023-03-14:
> We have obviously worked around this for the specific camera and it seems that Ximea have gone some way towards fixing it in their code which seems to the root cause of the issue so I think we should mark this as closed.

For the record, in the case of `"XI_TEMP_IMAGE_SENSOR_DIE_RAW"` we didn't work around anything and as far as we know Ximea hasn't fixed the issue. We just return a non-sense value of `0.0`.