python-microscope/microscope #171: Enable logic is wrong on CoolLED

iandobbie, 2020-10-27:
Not sure exactly how we want to code this but when connected to cockpit enabling a line on the CoolLED turns the light on permanently, then to get hardware control we need to send one trigger which then turns the LED off until the hardware line is high. We need to have an enabled mode where the hardware signal will turn the LED on but it is not on permanently immediately. This is similar to the issue we have with the toptica iBeam laser module. Fort hardware control we need a m ode where the light is read for a trigger but NOT emitting until the HW trigger arrives.

carandraug, 2020-10-29:
I think this is a specific case of #159 ?

carandraug, 2021-01-13:
All light sources now implement the `TriggerTargetMixIn` interface. I have now added support for hardware triggers in CoolLED too. Basically, one needs to specify the trigger type before enable, like so:

```
coolled = CoolLED(...)
channel_b = coolled.devices['B']

# for software triggers, calling enable will start emitting light:
channel_b.set_trigger(TriggerType.SOFTWARE, TriggerMode.BULB)
channel_b.enable()

# for hardware trigger, calling enable with "select" it, meaning channel is waiting for hardware triggers
channel_b.set_trigger(TriggerType.HIGH, TriggerMode.BULB)
channel_b.enable()

# disable unselects it, meaning that it will not emit light whatever happens
channel.disable()
```

The CoolLED has one issue, though, as described in https://github.com/python-microscope/vendor-issues/issues/9 . Basically, if a channel receives a TTL signal when it is "on", it reverts back to being "off", i.e., it will only emit light on receipt of further TTL signals, and querying its state reports being "on" (incorrectly).

closed 2021-01-13 by carandraug