python-microscope/microscope #172: Setting too large an exposure on Ximea camera.

iandobbie, 2020-10-30:
The Ximea xiQ has a maximum exposure time of 1s. If I use cockpit to try and set the exposure time to 2s it fails, and leaves the exposure time as previously set. 

Microscope logs the following error

xiAPI: XIA(f800):xiSetParam exposure out of range
2020-10-30 16:22:32,017:XimeaCamera (microscope.cameras.ximea):DEBUG:PID 1339: set_exposure_time exception: ERROR 11: Invalid arguments supplied

carandraug, 2020-10-30:
What xiapi version are you running? I have a XiC camera here and if I set an exposure time higher than the maximum value, then xiapi clips it (as expected). 

```
$ python3 -c 'import ximea; print(ximea.__version__);'
4.21.18
```

It might be a ximea version problem of maybe different cameras have behaviour.

iandobbie, 2020-10-30:
Sorry I have left the lab. I will try to ssh in later tonight.

Ian

carandraug, 2020-10-31:
I got the two cameras the two cameras and confirm that the issue that the different cameras have different behaviours. the xiQ will error when setting the exposure time too high, while the xiC will clip the value (kinda, it will accept the above max value and return that value, but once the camera is enabled, it will clip).  I have now reported this to ximea (ticket number 56174 on their internal private ticket system).

carandraug, 2020-10-31:
In theory we could just query the range, but ximea's latest update earlier update, broke it for the xiC. With version "4.21.18", the maximum exposure is "30000000.0", but with version "4.21.19", the maximum exposure is "7.411626765319505e-38".

iandobbie, 2020-11-02:
Sounds like an unsigned int recast as a signed float.