python-microscope/microscope #173: PrimeBSI pvcam image data seems to be limited to 12 bits.

iandobbie, 2020-11-12:
the prime BSI (using the pvcam driver) has two readout modes, 11 bit at 200 MHz or 16 bit at 100MHz. When In 11 bit mode the data tops out at 2047 but in 16 bit mode the data tops out at 4095, so the top 4 bits seem to have been lost.

iandobbie, 2020-12-07:
There was something weird with the pvcam install, or the camera config. It later appeared to have the correct 16 bit options. I suspect this is a config issue, or some kind of setting I inadvertently enabled. I suspect user error and as I dont have the camera at the moment I am closing this issue.

closed 2020-12-07 by iandobbie

nkhdiscovery, 2021-05-04:
@iandobbie How did you get it work with linux in the first place? I have a PrimeBSI too, when I connect it it shows up under `lsusb`, I have installed pvcam drivers, and python-microscope successfully imports the pvcam module, but when creating a camera object like `camera = pvcam.PVCamera()`, and trying to call `camera.enable()`, it says No cameras found (Camera is connected via USB port). I had trouble understanding how the camera works under linux, should it show up under `/dev/` for example? I am looking to find a way to make sure the camera is being recognized by my OS (Ubuntu 20.04) and then see why python-microscope cannot enable it. 

Any help would be appreciated, I did not found any clear tutorial anywhere. If you have found one let me know, and if not, this will be the document for future people who face trouble with their pvcam supported device.

iandobbie, 2021-06-07:
I have never used the primeBSI on linux, just windows. It turned out our primeBSI had a more fundamental problem and had to go back to the manufacture for repair. You should be able to to do

 import microscope
 import microscope.cameras.pvcam as pvcam
 cam=pvcam.PVCamera()
 cam.enable()

I don't have a primeBSI at hand here but I will try tomorrow, however I don't have a suitably setup linux machine and photometrics have stopped supporting macos. I will try to install it on a raspberry Pi and see if I can get that to work.

carandraug, 2021-06-07:
Hi @nkhdiscovery does the PVCam SDK comes with any "supported" program to connect their cameras?  If so, is the camera recognized there?

Also, since this question is unrelated to the issue here (PrimeBSI previously being limited to 12bit), could you please ask on [image.sc](https://forum.image.sc/tag/python-microscope) where there's likely to be more people with experience with the different hardware?