python-microscope/microscope #174: PrimeBSI doesn't work in experiments.

iandobbie, 2020-11-13:
The prime totally fails to work in experiments. There is zero lag between the end of one frame and the start of the next, but there maybe other issues as well.

carandraug, 2020-11-13:
Microscope does not have experiments and does not handle time between frames. I guess this comes from how cockpit uses it. So is the problem that that the reported cycle time is wrong? I was under the impression that the PrimeBSI was being used successfully on the Aurox system, but I haven't been involved in that system or anything to do with the PrimeBSI. Is this a regression then? What exactly no longer works?

iandobbie, 2020-11-13:
Nick reminded me that it was working fine on the Aurox several months ago. I am very perplexed to be honest but I couldnt get it to run experiments today. I gave up. I suspect the camera got seriously confused, crashed overheated or similar. Even in hardware trigger mode it was triggering continually. It also seemed to have 12 bit modes some time and 16 bit modes at other times. Not a Friday evening problem. I will get back to it mid week next week due to the micron course.

iandobbie, 2021-03-31:
I suspect this may be due to the broken camera that we sent back to photometrics. I am closing this as a suspected hardware issue.

closed 2021-03-31 by iandobbie